phil-barrett/grblHAL-teensy-4.x #1: Eagle files

ToastyStoemp, 2020-11-27:
Hey,
Are you planning on making the eagle files opensource for version 2.09 too?

klutchplate, 2021-03-27:
Hi.
I wanted to order 4 boards on Tindie, one for our large laser engraver, one for our current CNC router, one for our upcoming plasma router and one for our large CNC router (3m x 2m with rotary axis).
It turns out Tindie does not ship to, my country.
I was about to order 5 boards on JLCPCB, but noticed some missing holes and realized it's not v2.09.
Checking your files again and the file teensy 4.1x209 only contains the schematic that somehow gives errors when paired with the 4.1x207 files.
Will you be making v2.09 files available?

phil-barrett, 2021-03-27:
what country are you in?

klutchplate, 2021-03-28:
Johannesburg, South Africa.

phil-barrett, 2021-03-28:
Hi, I have enabled shipping to SA.  

I will look at providing the brd file but have a lot on on my plate right now.

klutchplate, 2021-03-29:
Great.

Thank you very much.

heripriyanto, 2021-07-03:
Hai Phil,saya juga ingin membeli v.209...tolong aktifkan pengiriman ke Indonesia.terimakasih