phil-barrett/grblHAL-teensy-4.x #2: Laser raster speed

theomajigga, 2020-12-07:
Phil, great work on this. I know very little when it comes to this type of ports, but my realization of how cheap M4 boards have become, are leading me down the path to upgrade me co2 laser to a proper (freshly written) 32-bit port of GRBL.

First: Sorry for an issue report, I know this isn't the proper route.

Second: I am currently running GRBL-LPC for my laser, I get stuttering on high dpi, fast movements (254dpi/300mm/s). I am hoping to get some increased speeds out of my machine, as it is capable of 500mm/s (with out imploding :) ) 

Per Oz of the project Lightburn: 

>  An image at 254 dpi means 0.1mm dots, or 10 dots per mm. If you run 100 mm/sec, that’s 1000 dots per second, and each dot takes a gcode instruction to produce.
> 
> Stock Smoothieware hits 800 instructions/second, or 80mm/sec with 254dpi.
> 8-bit GRBL is about 400, or 40mm/sec with 254 dpi
> 
> If you’re running GRBL-LPC or the new cluster Smoothieware (available on C3D’s forum) you can hit about 250mm/sec at 254 dpi.

After doing extensive research, it led me to your project, and some of the work you did with the gnea fork.

I guess, my question is, would upgrading to the teensy 4.x and the board you have produced on tindie, get me the speeds I am looking for?

Thank you!

phil-barrett, 2020-12-07:
No problem posting in Issues. 

I don't have solid answers for you but am pretty confident grblHAL on the Teensy 4.1 (600 MHz iMXRT1062 ARM M7) will outperform the LPC.  

Some data - the old Grbl/gnea running on an ATMega 328P was limited to a step rate of about 30 kHz.  In my latest testing, I have seen step rates of up to 400 kHz on the T4.1/grblHAL.  I did a [blog about it here](https://www.grbl.org/single-post/how-fast-can-it-go). If the min step pulse width wasn't limited to 2 uS, I'm sure it could go faster.  Depending on the gearing and microstepping of your machine, you can get significantly higher speeds than 6000 mm/min (100 mm/s). 

grblHAL allows building with a much larger input buffer and Terje's sender supports something called aggressive buffering which allows filling the input buffer so the communications link tends not to be the bottleneck.

You might also want to ask your question on the grblHAL issues list (they won't mind it at all). And, in fact Terje (Mr grblHAL) is very interested in laser support. He's looking at support PPI mode, for example.

theomajigga, 2020-12-16:
Awesome Phil, I wish my controller was more accessible. HAHA!

I think I will order one of your boards, not much I can break on my machine. 

A couple notes:

1. Users that would move over to this system, will NEED to be able to use Lightburn, it is hands down the best piece of software. The aggressive buffering, I believe is used per comments in https://forum.lightburnsoftware.com/t/gcode-controller-transfer-mode/26006/14 I am willing to test this out, after the holiday season! Orders coming in ;)

Is there anything that I would be losing when moving over to this new board? I was reading through https://github.com/terjeio/grblHAL/tree/master/drivers and it seems that the GrblHAL operates the same(ish) on both the LPC1768 and the teensy. I guess, my question is, are there any lost features on GrblHAL vs the last GrblLPC (Just a port from Grbl 1.1f I believe), or will I at minimum gain features? 

After reading through those drivers, I got more confused, but I believe that was stuff that Terje was just working on implement?

Thank you!

Cncninja, 2021-01-22:
Was there any resolve with laser use, I bought the new rev board but in the prebuilt firmware, i don't see a laser option also what board output would I use for my CO2 laser the PWM? or one of the I/Os? I would love to know how you connected yours up and the firmware used. 

Thanks

phil-barrett, 2021-01-22:
I replied to your question on the grblHAL discussion section.  You should be good to go.

theomajigga, 2021-03-15:
Teensy is in, and Phil, your board should be arriving today, will be testing quite a bit with lightburn and the latest ""3 axis usb no opts" - 3 Axis, Aux Relays" release.

Hoping to have some good info and some deep testing in in the next few days.

theomajigga, 2021-03-17:
@Cncninja It is way faster, I used PWM of spindle and V+ (5V) two spots down. Using the increased planned buffer, running into a couple of issues that I am working on ironing out, but it works beautifully.

theomajigga, 2021-03-17:
TL is what my input is on my laser power source.