phil-barrett/grblHAL-teensy-4.x #3: Pre-sales questions

frdfsnlght, 2020-12-09:
I'm getting ready to start a couple of projects (MPCNC and a mini-lathe CNC conversion). I have some questions:

1. The Tindie store _just_ ran out of the boards. When do you expect the next batch to be available?
1. I've been reading about some of the work on grblHAL about lathe spindle sync but I'm not clear on where it currently stands. It sounds like this board will support sync through the ST0-3 inputs (index and encoder). I'm wondering if that's still the plan or do you foresee a hardware change in the future?
1. Will this board support a MPCNC auto-squaring with dual end stops? Assuming dual NC limit switches on both X and Y, can I just wire each pair in series and connect to the X and Y limit inputs? And the steppers would be connected to X/A for the X axis, and Y/B for the Y axis?
1. Does the e-stop switch connect to the Halt input? And a momentary, NO switch to the Fd//Hld input? What's the practical difference between these signals? Once halted, can the program be resumed by a momentary NO switch on the Cyc/St input in either case?
1. What's the purpose of the Probe input? Is that a switch input to trigger a probe cycle, or is that the probe input itself? If it's not the probe input itself, where does the probe input connect? On a "typical" MPCNC, probe input connected to the Z-min limit, but the limit inputs on this board don't distinguish between min and max.
1. I'm planning on someday adding a MPG, probably building it myself from buttons, an encoder, and microcontroller (perhaps a Teensy 2.x). There was some discussion of being able to connect something like that via USB to the Teensy 4.1's USB host port. Would you anticipate any problems doing that with this board? I assume there wouldn't need to be any hardware changes to the board?

Thanks for any help that can be provided. I apologize if this isn't the correct place to post questions like these.

frdfsnlght, 2020-12-09:
OK, question 1 is answered. Tindie _just_ notified me the boards are back in stock. Yay!

phil-barrett, 2020-12-09:
Yeah, I got more stock just in the nick of time!  This is a good place to ask the questions though the brand new discussions feature looks like it might be better. I think it was introduced , literally, minutes after you posted here!  These are really good questions and worthy of a FAQ. 

2) I'm not a lathe guy but have been following the discussion closely.  Yes, ST0-3 are intended for quad input and spindle sync. To be honest, when I laid out the current board, I was unaware of the need for spindle sync.  I have a new board I am working on (Pro version) that changes the ST pin assignments. Not sure when that will be released. You should ask your lathe questions on the [grblHAL ](https://github.com/terjeio/grblHAL) discussions section - several lathe people are active over there.

3) Yes, auto squaring is supported though it has it's limitations.  There is at least one thread on autosquaring in the grblHAL issues section.

4) Yes, EStop should connect to halt. You might find [this blog](https://www.grbl.org/single-post/estop-doing-that) I did on that topic useful. Halt/EStop terminates a GCode program though retains X, Y and Z zeros. My old Grbl/gnea board did not always retain the zeros on EStop.. Feed Hold pauses GCode program execution and Cycle Start allows resuming it.  I recommend a latching switch for EStop and momentary switches for Feed Hold and Cycle Start.  grblHAL allows you to use NO or NC for any of these inputs. The blog mentioned above goes into this in more detail

5) Probe is a very useful input.  In it's simplest form, it is used with a touch plate to find the height of your workpiece. The G38.x commands are used to the stop lowering Z when a probe signal is seen and then you would set the Z height via G92.  A number of GCode senders use the probe input to things like finding the edge of a workpiece, inside center finding, outside center finding and even height mapping of an uneven surface (and converting a GCode program to follow the height map - great for PCB machining and etching). hmmm, I need to do a blog post on that.

6) yes, it is my intent to support the Host USB interface for MPG/Pendants.  The new Pro board will have a connector on the breakout board.

frdfsnlght, 2020-12-09:
Thanks for the response! I'll do some searching for the autosquaring stuff in the grblHAL repo.

The new Pro board sounds interesting. Do you have any expectation when it might be available? Not looking for a guaranteed date, just trying to get an idea of timing for my projects. Some Pro questions:

1. Other than needing a different firmware build since the pin assignments are changing, would you expect spindle sync to work with the ST inputs on the current (non-Pro) board?
1. I assume the USB host connector will just be a standard USB port (micro?), routed directly to the pins on the Teensy on the new board? I normally extend all the ports (USB/Ethernet) to a back panel in the controller case, so connecting directly to the header pins on the Teensy isn't a problem for me. Do you anticipate any magic added to the new board that might make me 
 rethink that?

Thanks for the help!

phil-barrett, 2020-12-09:
Timing wise, I have the design mostly done and a few test boards built up. I have been making a few changes - supporting 12V limit/control inputs rather than 5V, for example.  This board will have pluggable screw terminals and I am playing with spacing to make it easier to have a good quality build (seen too many Chinese boards with terrible connector alignment).

I do not know the answer to the spindle sync question. You should ask that in the grblHAL group.

I am using a USB A female connector on the Pro board as most MPG/Pendants use a USB A Male connector.  I expect people will use a short "bulkhead" cable between the board and the case so  you can adapt just about any connector via that.  If you use the Teensy 4.1 header, make sure the wires aren't too long.  I've tested with 6" and that seems to work ok.

frdfsnlght, 2020-12-09:
Hm. The Pro board sounds appealing. Maybe I'll wait a while longer. I'm in no rush, just trying to plan out some winter projects.

frdfsnlght, 2020-12-14:
So... I don't know if you're willing to talk too much about the upcoming pro board, but I'd like to get all the info I can during my design phase.

* What are the board dimensions?
* Are the connectors going to be arranged around the 3 outside edges like the current board?
* Other than the pluggable terminals, what new "pro" features are you planning?
* Any pricing info (even approximate)

I'm mostly after info that differs from the current board in a physical sense. I'd like to make sure there's room in my box for the new board and room for buttons/switches to attach to new, interesting features, if any.

I'd appreciate anything you could give me. Of course, this might not be the best issue to talk about all that. If you're willing to share, you could create a new issue describing the plans, or contact me directly if you don't want it public yet. I promise I won't tell anyone if that's what you want.

phil-barrett, 2020-12-14:
This thread <https://github.com/terjeio/grblHAL/discussions/130> in the
grblHAL discussions section is a good place to start.

The target board size is 100x100mm (4x4) but plan on some overhang of the
pluggable screw terminals.  Also, I generally try to have at least 2 inches
of free space around the board - makes wiring a lot easier.  I am switching
to a vertical entry Ethernet  connector so that reduces some of the space
needs around the board.

frdfsnlght, 2021-01-21:
I'm reviving this issue just to get an update on the new pro board. How's it coming along? Any ETA yet?

Is there any other information about the board available? At the moment, I'm mostly interested in where the various inputs and outputs are located (i.e., which edges are the limits switches on, the ethernet port, motor signals, etc.).

My CNC router build is still in progress, but I'm at the point where I'm putting together the "controller box" and I'd like to get an idea of where to route the various wires inside so they're ready to be connected to your board. A labeled image or PDF of the board would be great. Of course, all that wiring won't take long, so I'm also interested in when the new board will be available because I'm running out of things to do :)

phil-barrett, 2021-01-21:
Current Pro board status - received test PCBs.  Debugging isolated spindle support looks promising but need to do a full work-up with the Oscope.  Found problem with ethernet. probably need another turn of the PCB. 

Currently all the I/O connectors are in the same place.  Board is bigger - approx 100mm x 100mm. Power connectors are moved  a small amount and a new power connector has been added for isolated 12V for the inputs (end stops and control switches) in the lower left.  I2C header has moved to the right about 30mm Vac/DC switch header has moved.  Ethernet connector is now vertical entry.  New USB Host connector is also vertical and next to ethernet connector.

I haven't yet sized the footprint of the board with wiring but it will be larger for 2 reasons: the board is larger by about 20mm) and the pluggable screw terminals add around 10mm on all sides. So, with ferrules and wires, a reasonable guess is 150mm x 150mm of space is needed. For Imperial types, 6" x 6" is a safe bet.

frdfsnlght, 2021-01-21:
Great info! Thanks! I should have room for the new board.

The USB host port, what kind of port is that? I plan on ordering some port-to-panel extenders. I know the Teensy is a micro USB to USB-B at the panel, and the ethernet will be a RJ45 to RJ45 at the panel. I want the USB host to be a USB-A at the panel, but what do I need at the other end (on your board)?

Where do you go for your PCBs? I'm sure your runs are larger than what I normally need when I order PCBs, but I'm always open to new options.

phil-barrett, 2021-01-21:
The Host port isn't currently supported in grblHAL but it is just a USB A female. So USB A male to USB Female.  The idea is you could plug in a USB based MPG/Pendant or similar device.

I use a variety of board houses: in no particular order JLCPCB, ALLPCB, OSHPark (small stuff)., PCBWAY  Kicking the tires on a couple of others.  PCBShopper is a good resource for evaluating options.

frdfsnlght, 2021-01-21:
Thanks for the info. A MPG/pendant is just what I want the host port for. I understand grblHAL doesn't support it... yet. That may be something I try to get my hands dirty with eventually.

I typically am doing "small stuff" (projects just for me) and frequently use OSHPark. In fact, I'm waiting on some very small boards from them now. I designed a very simple "limit switch aggregator" to connect multiple NC proximity sensors (really just 2) to a single limit switch input on your board.

phil-barrett, 2021-01-21:
Yeah, I use OSHPark a lot for little stuff. Wish they had frequent flier miles - I get at least one package from them each week.

Does your aggregator do both series and parallel (ie. NC and NO)?

frdfsnlght, 2021-01-21:
No, it's designed for NC only. NO switches can just be wired in parallel. I wanted to keep NC proximity switches for all the good reasons, but in order for 2 or more to share the same input pin, you have to wire them weird (to me) and that type of wiring won't work in my cabling scheme. So I found/designed/tweaked a simple circuit made of 2 diodes, 1 resistor, and an NPN transistor. You can add more switches by adding a diode for each one, but I only need 2 for each input (min/max). The output of the aggregator is NO, but since it's all solid state and contained inside a box and always hooked up to the controller, I figured that short path was OK to be NO.

frdfsnlght, 2021-01-22:
Ha. I just realized my new boards won't work the way I intended. I need to make some changes and re-order. Such is life.

frdfsnlght, 2021-01-22:
Is there any way you could provide a snippet of the schematic for the new board around the limit switch inputs? I believe you said you were adding isolated 12V for the inputs. I'd like to see how you're doing that to make sure my new aggregators will work.

phil-barrett, 2021-01-22:
Here is the schematic for isolated inputs. They are pretty much the same as with the original board but with an external voltage supply and running at 12V
[image]

frdfsnlght, 2021-01-22:
Great, thanks! Is the order of the inputs in the schematic the same as on the board edge?

phil-barrett, 2021-01-22:
Currently it is.  Though, I am not committing to that exact layout yet.  I may yet need to move things around.

frdfsnlght, 2021-01-24:
I'm in the process of redesigning my aggregator board. It looks like you set the current of the opto's to around 6.8mA. Do you think I could get away with adding an SMD LED to the outputs of my board (inputs of yours)? I'm looking at LED's with a V<sub>f</sub> of 1.7V and I<sub>f</sub> of 20mA. That brings the current for the optos down to around 5mA, which, if I'm reading the EL357N datasheet correctly, should still work. Do you concur?

phil-barrett, 2021-01-24:
You are saying that you want to add an LED in series in addition to driving the opto input.  Just making sure I understand the context.

On my board I use 330 ohm for opto inputs so the I<sub>f</sub< is around 11 mA.  You do need to look at the min and max also, to avoid any edge issues, including min and max on resistor tolerances and various selection bins. The EL357s work pretty well down to less than half that I<sub>f</sub> but I like the higher current for noise immunity, especially with NC switches.

If you just put your 1.7 V<sub>f</sub> LEDs in series with the inputs it would bring the opto's I<sub>f</sub> down to 3.3 mA which is pretty low. It might work ok but it's hard to say - the datasheet implies a normalized collector current of around .5mA in that situation which would be on the low side. It would take a fair amount of analysis and testing to know if it is 100% safe.

If I were designing an aggregator I would probably take a different route to avoid affecting the opto inputs (you never know what some cheapo designer has done on the other side...). Use a small line driver or inverter to drive LED and thus drawn negligible current.  I like the LVC1G17 - single input Schmidt trigger with lots of current drive. $0.0574 Q/1 from LCSC. You could probably live with just it, a resistor and the LED. Takes up a tiny amount of space. Hand solderable - you might need a steady hand and a little magnification.  Though, some sort of hobbyist reflow would be easier.

frdfsnlght, 2021-01-24:
I'm a little confused now. Your older board (non-pro) used 330 resistors, but the schematic you posted above uses 1.5K, which would yield 7.2mA. Or did you intend a smaller R value to bring I<sub>f</sub> closer to the 11mA of the original board? With the 1.5k and a 1.7 V<sub>f</sub> LED, wouldn't the current then drop to 6mA? I understand that's lower than you might would want and parts tolerances could screw me. I first just want to make sure I'm understanding the math. If you replace the 1.5K with a 1K, you'd have closer to 10.8mA without the extra LED and 9.1mA with. Or am I doing something wrong?

I totally understand about not wanting to effect the opto inputs, but I'm not sure your Schmidt trigger chip works either. The proximity switches are open collector with a pullup to the supply voltage, which will be 12V.

Here's my schematic so far:
[image]

I took inspiration from your design and decided to use the same optos. This circuit allows me to directly connect the cables to each axis (XYZ), each of which have a min/max switch, and the A axis for an extra switch for auto-squaring. The probe input gets the same treatment. The dual sensor axis just wire the optos in series and maintains a "NC" path all the way to your board. It also makes a bunch of my wiring easier.

The idea for the LEDs would be to insert them in series with the my opto LEDs. I can change my resistor values to keep the current up. The issue is that I can't change your resistor values, so I'm trying to figure out the feasibility adding LEDs on my outputs (your inputs).

phil-barrett, 2021-01-24:
Sorry, I didn't realize you were talking about the 12V input version.  You are right the +5V logic doesn't apply in that case.  sorry for the confusion.

For what it's worth, replacing the resistors for the optos isn't that hard because I will be using 805s (higher dissipation) and they are pretty easy.  Plus, there is a fair amount of room available. 

Of course, I have to finish the board and the isolated 0-10v section still has issues. sigh...

frdfsnlght, 2021-01-25:
I use 805 myself, but that's just because I'm getting old and manually placing component smaller than that doesn't work for me. Oh, and I do have a reflow oven.

But is it your intent to use 1.5K resistors on the 12V version? I picked 1K for my board because it brings to current back up to almost 11mA.

phil-barrett, 2021-01-25:
Yes. I was concerned about power dissipation as the 603 packages are typically rated at 100 mW and 1K would be a bit above 116mW with a typical V<sub>f</sub> of 1.2V.  That's why I went with an 805 package which has 125 mW dissipation.  But, that is awfully close and there is no V<sub>f</sub> minimum listed for the EL357.  1V would put it right at the limit and I am moderately conservative about this sort of thing. 1.5K is a reasonable trade-off. Plus, 12V NC switches would take a pretty big ESD hit to flip the opto.

frdfsnlght, 2021-01-27:
Power dissipation is the thing I almost always forget about. Circuit design is not something I do all the time so I get rusty.

So, I just ordered some 1210 resistors (1/2 watt, good to have in stock anyway). Perhaps I'll use those with some LEDs on my inputs and leave any LEDs off the outputs (your inputs), unless you decide to bump up the resistor size on your board. Having indicators on the inputs is useful since the sensors are scattered on the machine, so this way I can get a sense of what all of them are doing by looking in one place.

phil-barrett, 2021-01-27:
You will be hard pressed to blow a 1210! 

I agree, LEDs to indicate the state of the inputs and outputs is worthwhile.  I had originally had a number of them but it made the board a lot bigger for a number of reasons.  I hope to add them in as my boards evolve.

frdfsnlght, 2021-01-30:
So I must have been having a senior moment. I just realized there's no reason to have the LEDs in series with the optos. Since the sensors and optos are open collector, I can just connect the LEDs between 12V and the collectors with a resistor. There's no extra load through the optos and I use whatever resistor values I need for the LEDs. I just have to make sure I don't exceed the current the optos (and sensors) can sink, which shouldn't be a problem (I think it's 50mA?).

phil-barrett, 2021-01-30:
Actually, LED in series with the Opto emitters (basically LEDs too) is not a dumb idea at all but it does depend on the source voltage (5V in this case).  You would have to subtract the 2 V<sub>f</sub>s from 5 and then divide by the resistor to get the current.  

One of my ideas for LED indicators on the base breakout board was exactly that.  This would be in a 12V system.

frdfsnlght, 2021-02-06:
More questions about the upcoming pro board...

1. I believe you said you're using pluggable terminals. Are they vertical or horizontal entry? What pitch (3.5, 3.81, ...)? Do you know if the PCB footprint is different for the vertical vs horizontal?
2. You said the new board is 100x100mm. Is that still true? What's the mounting hole spacing? I assume the holes are in the four corners?

frdfsnlght, 2021-02-07:
I've just had an idea... I'm getting close to needing a controller for my new router. I'm not sure I want to wait for the pro board (I'll definitely want one eventually, no pressure :)). I'm trying to figure out how to make my aggregator work with your current board at 5V. I'll still supply the sensors with 12V, but that means the reverse voltage on your optos will exceed the 6V max. So, I have some SS34 schottky diodes (0.5V forward drop, 40V max reverse) I'm thinking of putting inline with my outputs (inline with your inputs) to prevent my 12V getting to your optos and frying them. I don't think the small 0.5V drop will cause any problems triggering your optos. And they'll allow my board to also work when I upgrade to the pro board. What do you think?

And on a related note, are you sure about the 1.5K resistors on the inputs in your pro board? If your target current through your optos is 10mA (like on the current board), don't you want 1K? Or has your thinking changed?

Here's what I have now:
[image]

phil-barrett, 2021-02-07:
> I believe you said you're using pluggable terminals. Are they vertical or horizontal entry? What pitch (3.5, 3.81, ...)? Do you know if the PCB footprint is different for the vertical vs horizontal?

3.5mm.  Horizontal for now.  Vertical is interesting to minimize the overall footprint of the board and wiring.
> You said the new board is 100x100mm. Is that still true? What's the mounting hole spacing? I assume the holes are in the four corners?

Currently 100x100.  The mounting holes are 3.2mm and spaced 96 mm apart in a square. Please don't make anything based on these numbers, I can't guaranty they won't change.

I don't think I asked - what are your limit sensors?  Open collector or totem pole (aka push/pull)? Do you have specs? Some of those sensors run at 5V even though spec'd for 6V min.  I have 2 sets like that.  It has some bearing on your design.

Your design presumes NC sensors, as I am sure you know. You could solve the "feeding 12V to the board" problem by running LimA and Probe through optos and running all the optos at 5V.

I will revisit the resistor values when doing the first pre-production board.  1K would be perfectly fine and have a higher current transfer.

frdfsnlght, 2021-02-07:
I'm going to put vertical entry pluggables on my board for that reason; less space needed around the board.

The only thing I might make that depends on the mounting hole spacing would be 3D printed, so I can always make something different if I need it. I'm thinking of printing a "tray" to hold your board (both the current one and the pro board) next to my own board, then just mount the tray in the box.

My limit sensors are LJ12A3-4-Z/AX, NPN NC, 6-36V. I believe they're open collector with a built-in 10K pullup to the supply voltage (at least that's how they act). I haven't tried them at 5V. If they work at 5V, I could eliminate the diodes, but then the resistor values for the LEDs and optos are wrong. I'd rather make my board work at 12V all the time and feed into a 5V or 12V controller. When working with a 5V controller (like your current offering), I wouldn't use the VCONT terminals to feed the 12V into the controller. That's only for the isolated input voltage on the pro board.

phil-barrett, 2021-02-08:
By the way, vertical Pluggable Screw Terminals have a different footprint than horizontal ones. Vertical PSTs are much closer to non-pluggable screw terminals (like on the current board).  Slightly different spacing, though - approx 7.9mm vs 7.

frdfsnlght, 2021-02-08:
Is the plan to offer the pro board in an "unkit" version where I have to buy and solder all the terminals? Or as a complete board? I'd prefer to have vertical PSTs for the space savings. I don't mind the soldering, but I'd rather not un-solder.

What Mouser/Digi-Key part number(s) are you using for the 3.5mm PSTs?

phil-barrett, 2021-02-08:
Using LCSC as the disti. Here are the base parts.
Horizontal:
https://lcsc.com/product-detail/Pluggable-System-Terminal-Block_JILN-JL15EDGRC-35002G01_C409115.html
https://lcsc.com/product-detail/Pluggable-System-Terminal-Block_JILN-JL15EDGRC-35002G01_C409117.html

Vertical:
https://lcsc.com/product-detail/Pluggable-System-Terminal-Block_JILN-JL15EDGVC-35002G01_C394557.html
https://lcsc.com/product-detail/Pluggable-System-Terminal-Block_JILN-JL15EDGVC-35004G01_C400551.html

frdfsnlght, 2021-02-09:
Thanks for the links. But I can't find the terminals that plug into those sockets. Do you have a link/PN for those? There are other manufactures that make the plugs, but I can't find the sockets for them. How interchangeable are these things between manufacturers?

It's clear from the datasheets for the parts you linked to the footprints are different between the horizontal and vertical sockets, which isn't surprising. Are you placing the sockets on the board so that either horizontal or vertical sockets could be used? If you're not, could you please do that? Pretty please?

You didn't answer the question about the new board being an "unkit". I *really* want vertical entry terminals the more I think about it. While not ideal, I could de-solder horizontal sockets and replace them with vertical (assuming there's room on the board). More ideal would be for the sockets to not be soldered on the board as purchased.

Any thoughts? Sorry for so many questions, but I'm kinda' a detail oriented person. I'll take any info you can provide.

phil-barrett, 2021-02-09:
LCSC doesn't have the most user friendly site.

https://lcsc.com/product-detail/Pluggable-System-Terminal-Block_JILN-JL15EDGK-35004G01_C393014.html
https://lcsc.com/product-detail/Pluggable-System-Terminal-Block_JILN-JL15EDGK-35002G01_C393013.html

There are other manufacturers. To find all of the pluggable screw terminals, go to All Products/Connectors/Pluggable System Terminal Block.

Then in the Package Section, multiselect 2 items: P=3.5mm and Through Hole, P=3.5mm.  Then hit apply If you select things from other sections that seem obvious (say pitch, 3.5mm), you get nothing close.  Took a while to figure that out.

I will probably offer an unkit first.

frdfsnlght, 2021-02-09:
You can say that again. I'm familiar with parametric search, but their site is wonky. Thanks for the links (and the tips).

frdfsnlght, 2021-02-14:
Some more questions to annoy you.

1. Your v2 boards are out-of-stock on Tindie. Any idea when you might get more in?
2. To use Ethernet on the v2 board, you need to have the 2mm header soldered on the bottom of the Teensy. If you want to socket the Teensy (I know, not recommended to do when using Ethernet, but this might just be temporary) you also need the 2mm socket soldered to the controller board. The 2mm header need to be long enough to reach given the height of the 0.1" headers on the sides of the Teensy. You've provided Mouser/Digikey parts numbers for the 2mm header and socket. Are their any parts numbers for LCSC? I'm getting ready to place an order on LCSC and I haven't been able to find those 2mm header/socket on LCSC (I can't find headers long enough and the sockets are a little too wide/long). I'd rather not have to place a Mouser order just for that header/socket.

Thanks again.

gipsy58, 2021-02-14:
Hi Phill

going through the documentation and some halgrbl code didn't help help me too much. Are you going to support A and B input from a  quadrature encoder?

rgrds

phil-barrett, 2021-02-14:
Tab, Hoping to get more boards soon.  I tried to get an order in before Chinese New Year kicked in but missed by a little that turned into a lot, sigh.  Right now I have a batch of them finished but DHL doesn't seem to be picking up.  They have been awaiting shipment for 5 days now. Hopefully this is the week at least some of them get back to work.

As to LCSC part numbers for the long 2mm 2x3 header.  I took a quick look and didn't find.  The obvious params of pins=6, rows=2, pitch=2mm gets sockets and pin headers, but no long pin headers. They may be there but, given the mess of their parametric searching, it will take more effort. You could try backing off each parameter - I'd start with pitch which seems to be unreliable in their search.

phil-barrett, 2021-02-14:
@gipsy58,

Yes, it is supported.  You can select it in my_machine by uncommenting line 58 (+/- one or two):
`#define QEI_ENABLE         1 // Enable quadrature encoder interfaces. Max value is 1. Requires encoder plugin.
`
There are several builds for that in the pre-built firmware.

From the board map file (T41U5XBB_map.h), on the PCB, Pin A is ST1, Pin B is ST2 and the switch pin is ST3.  Good luck.

phil-barrett, 2021-02-14:
@frdfsnlght 

I just had an idea. Try stacking 2 sockets to get enough height. I don't have any spare sockets right now but eyeballing them, it looks pretty close. Electrically, not a great idea but if it works...  You may need to tin the 2nd socket's pins to make them larger if they are too loose. LCSC has several 2mm ones to chose from.

frdfsnlght, 2021-02-14:
Yeah, I tried a lot of things in the search and couldn't find them either. Was hoping you had found them once upon a time.

Stacking 2 sockets is a good idea. However, all the sockets on LCSC seem to be larger (wider/longer) than the ones you specified, by a fraction of millimeter. Looking at the Teensy, I'm not sure if those larger sockets fit right up against the 0.1" headers. I mean, they're cheap enough I could try it anyway. I suppose I could sand them down a little too.

I've got an Tindie alert setup for the v2 boards when they're back in stock. I'm getting a lot closer to needing *something* to drive my new router. I've been dragging my feet for a while now just waiting for the pro board, but it looks like that's going to be a longer than I'm willing to wait. I can always upgrade someday, maybe when I get my lathe CNC'ed.

phil-barrett, 2021-02-14:
Next time I place an LCSC order I will add samples of each of the sockets available and give them a test.  

I am making steady progress on the Pro board.  I have the isolated spindle stuff sorted out except for the optional isolate 5V to 12V converter testing (should be easy - FLW). I also added a 3.3V vs 5V selection for the UART and that needs to get built and tested. Progress has been slower than I wanted.

gipsy58, 2021-02-14:
Hey thanks Phil

I must have been blind. I knew about the ST1/2/3 but I just could not find them on the board. My lathe should be ready in about 2 weeks and I will order the board soon.

The pin connections are good enough.

rgrds

phil-barrett, 2021-02-19:
@frdfsnlght 

So, I was looking at the vertical vs horizontal pluggable terminals and had one of those "why didn't I think of that sooner" moments.  I could create a custom footprint that would allow either type to fit on the board.  The only trick is to get Kicad to allow me to have a pad with 2 holes in it. A first try didn't go well but there's got to be a way.

elfren, 2021-02-19:
I'm using clones of Phoenix pluggable terminals (1840382 and 1897115 are the right angle model numbers).  The only difference between the footprints for these and the vertical models is the vertical needs about 2mm more space towards the inside of the board.  The pin spacings are identical.

frdfsnlght, 2021-02-19:
I regularly have those moments. I'm glad you had that one. Vertical entry will make a big difference for my application.

I've successfully done multiple-of-the-same-pad in a custom Kicad footprint. If I remember correctly, just add 2 pads and give each pad the same number and it just works. Might have to route between them in Pcbnew, or maybe you can do that in the footprint too. It was a while ago I did that.

BTW, Tindie notified me of the v2 boards coming back in stock and I ordered one last night. No extra parts, just the BoB since I have an LCSC order coming with all the extras, including my vertical pluggable terminals for my own board.

phil-barrett, 2021-02-19:
@frdfsnlght  Thanks.  That did it.  Will update the design.  Sending off the "final proof" of the Pro board for manufacturing after that. Got all my last little touches - 3.3-5 UART option (w/level shifting) and RS485 signal pin. Tested isolated PWM & 0-10V up to 100K f<sub>PWM</sub> just fine and linearity looks good on 0-10V even at 100K.

Just got back from dropping your package in the mail.  Now it's in the hands of the Postal Gods.

phil-barrett, 2021-02-19:
> I'm using clones of Phoenix pluggable terminals (1840382 and 1897115 are the right angle model numbers). The only difference between the footprints for these and the vertical models is the vertical needs about 2mm more space towards the inside of the board. The pin spacings are identical.

Based on the datasheets for the ones I'm using (links in previous post), the total difference is 2.3mm and the distance between the two alternate pin locations is 1.85mm.  In my horizontal mount layout, the connectors overhang the board by 2mm so it will fit pretty well (what's 0.3mm between friends?). The V and H receptacles will pretty much occupy the same board space. 

The only concern I have is for when I have a full assembly run.  Very careful instructions will be needed to prevent assembly errors.