phil-barrett/grblHAL-teensy-4.x #4: Build from Source Instructions

jimbalny, 2021-01-02:
I tried going through the instructions for building grblHAL for your breakout board from source and am having some difficulty as there's no main folder in grbl-HALmaster/drivers/IMXRT1062/main. Are there any updated instructions for building from source? I'm trying to build from a more recent grblHAL version as to try to get homing working on my machine. Had a few issues where I lost my zero position after unplugging my board and I think if I'm able to home I could record my WCS just in case I leave my workpiece on the machine to finish later again.

phil-barrett, 2021-01-02:
Happy New Year Jim,

The grblHAL Wiki has the latest [build process](https://github.com/terjeio/grblHAL/wiki/Compiling-GrblHAL).  Be sure to get the networking library if you want to build with ethernet.

jimbalny, 2021-01-03:
Happy new year Phil!

oh ok thanks! I must have clicked through a link from somewhere else with
old information!

closed 2025-03-05 by phil-barrett