phil-barrett/grblHAL-teensy-4.x #6: YL620 and similar VFD spindles compatibility

garrett-rethink-contract, 2021-02-06:
Phil, have you personally tested the breakout board with a YL620 or other similar spindles by Huanyang. Or made by somebody...they don't really put a name on the device or the documentation.

The spindle accepts a 0-10V input, which they typically show being used with an external potentiometer wired between their 10V output, 0-10V input, and ground.

I find that when I try to control spindle speed this way, there is a nice big ground bounce that resets the Teensy.

I already had to isolate the EN and DIR outputs with external relays, because those inputs on the YL620 are NOT logic level, it wants to sink something like 15 volts through the FWD and REV inputs to XGND. However I'm a bit grumbly at the prospect of having to isolate the 0-10V output somehow, it's a bit more difficult of a problem.

Just wondering if you or anyone else ran into this and have some tips. Future PCB modifications might include an optoisolated section for the spindle; one could easily tie grounds together if they wanted, but the spindle enable, opamp, and opamp supply regulator could all live in an isolated island. PWM will go through isolation just fine.

macegr, 2021-02-06:
(ugh, sorry for the wrong account above. this is my usual account. I'm leaving it, and commenting here to make sure I see a response)

phil-barrett, 2021-02-06:
I have not tested it personally though I have several customers that use it as well as the PWM output with VFDs.  

I have a new board coming that provides Isolated spindle (en, dir, pwm and 0-10V).  I was hoping to have it available in Feb but Chinese New Year and some additional debugging makes that wishful thinking.

I am open to resolving this situation how ever you would like. Return for a refund?  Credit for a future board?  Let me know.

macegr, 2021-02-06:
No chance on returning it, it's built into the machine and I'll use it in whatever way works, even if it's manual speed control :) The price on Tindie was suspiciously low, and I'd be excited to buy the upgraded version later whenever you release it.

phil-barrett, 2021-02-06:
OK.  If you do buy the next one, remind me of this conversation.

macegr, 2021-02-06:
If you want, you could go ahead close this issue since it'll be resolved with future hardware. I consider this to be completely acceptable since we're buying into sort of a bleeding edge system, this is the early adopter tax. I'm happy with the board overall and I still have start/stop/fwd/rev control as-is. Speed control on a router is a once per job type of adjustment.

phil-barrett, 2021-02-06:
Thinking about this last night, have you tried a toroid on the cable to the VFD?

closed 2021-02-06 by phil-barrett

macegr, 2021-02-06:
I tried series resistance, zener clamp, a ferrite bead. It didn’t help the situation. When the grounds aren’t tied together I can see a 120V waveform across them, but it’s weak (meter impedance too low but scope sees it). It occurs to me that the Pi, breakout, and its supply are not directly tied to chassis ground, will try that too.

macegr, 2021-02-10:
I can confirm that if you bond the Teensy ground directly to earth ground (green wire), and made sure the VFD, Teensy, breakout, computer (Pi in my case), and 24V power supply are all bonded together and to earth, the Teensy does not reset when the VFD 0-10V lines are connected and spindle is activated. I have speed control! The isolated version still sounds like a good idea since some motor spike is being dumped somewhere into earth, but it's working now.

phil-barrett, 2021-02-10:
Glad to hear it. VFDs and spindles are nasty devices from the electronics point of view.  Not as bad as HF Start Plasma Cutters, though.