phil-barrett/grblHAL-teensy-4.x #7: Issue in laser mode, laser turning on after movement, can't disable

TheAnt1dot3, 2021-02-06:
Hi,

I'm using your breakout board as both a cnc controller and sometimes switch to laser mode for etching.
The etching itself works fine, but i'm having issues with the laser constantly enabled or flashing.
for example, i have unplugged my usb and shut down external 5V power to my board. when i power it on and plug the usb in, the laser is flashing  (at full power it seems so it burns a hole in my board).
if i move in the positve X direction, the laser shuts off, but when i move in the negative X direction, it goes to full power and remains on until i move it to the positive direction again.

here are my config settings: 
GrblHAL 1.1f ['$' for help]
client> $$
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=1 (Step direction invert, mask)
$4=7 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=510 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$14=14
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=0 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=25.000 (Homing locate feed rate, mm/min)
$25=500.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=1.000 (Homing switch pull-off distance, millimeters)
$28=0.100
$29=0
$30=1000. (Maximum spindle speed, RPM)
$31=0. (Minimum spindle speed, RPM)
$32=1 (Laser-mode enable, boolean)
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$37=0
$39=1
$40=0
$41=0
$42=2
$43=1
$44=0
$45=0
$46=0
$56=5.000
$57=100.000
$58=-5.000
$59=500.000
$60=0
$61=0
$62=0
$63=3
$64=0
$65=0
$100=6400.000 (X-axis travel resolution, step/mm)
$101=6400.000 (Y-axis travel resolution, step/mm)
$102=1607.000 (Z-axis travel resolution, step/mm)
$110=750.000 (X-axis maximum rate, mm/min)
$111=550.000 (Y-axis maximum rate, mm/min)
$112=600.000 (Z-axis maximum rate, mm/min)
$120=25.000 (X-axis acceleration, mm/sec^2)
$121=25.000 (Y-axis acceleration, mm/sec^2)
$122=200.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, millimeters)
$131=200.000 (Y-axis maximum travel, millimeters)
$132=200.000 (Z-axis maximum travel, millimeters)
ok
Below the commands being sent. X -100 triggers the laser to be on after about -20   X +100 seems to disable the laser around the same mark.

note that if the laser is in the negative X direction, i cannot disable the laser with M5S0 (if the laser is in the positive direction i can turn it off and on)

client> $X
ok
feeder> G91
ok
feeder> G0 Y10
feeder> G90
ok
ok
feeder> G91
ok
feeder> G0 Y10
feeder> G90
ok
ok
feeder> G91
ok
feeder> G0 Y10
feeder> G90
ok
ok
feeder> G91
ok
feeder> G0 X-100
feeder> G90
ok
ok
feeder> G91
ok
feeder> G0 X100
ok
feeder> G90
ok
feeder> G1F1
ok
feeder> M3S1000
ok
feeder> M5S0
ok
feeder> G91
ok
feeder> G0 X-100
feeder> G90
ok
ok
feeder> M5S0
ok
feeder> M5S0
ok
feeder> M5S0
ok
feeder> M5S0
ok
feeder> M5S0
ok
feeder> G91
ok
feeder> G0 X100
feeder> G90

the laser is wired to the +12v and G input and the signal is connected to the PWM port on the board.
i'm running around in circles trying to figure this out, so i'm reaching out for help here.

thanks,

Stef

phil-barrett, 2021-02-06:
First, off.  The PWM output is also use to drive the LED on the Teensy so you can disconnect your laser while testing this. The brightness of the LED is proportional to the S-parameter setting in GCode.

Sorry for the dumb questions.  I'm trying break this down a bit.  

What grblHAL version are you using?  Did you build it yourself? What sender are you using? 

Are you getting the laser powered on during rapid moves?  (G0)  Even following an M5?  By the way, M5S0 should be the same as M5 without the S-parameter.  M5 turns off Spindle/Laser PWM and Enable.

phil-barrett, 2021-02-06:
I am trying to reproduce the problem and am not seeing it.  I have a logic analyzer hooked up and any PWM output will get caught - nothing shows.

What is your laser HW and how is the board hooked up to it? Maybe some pictures of your wiring, if possible.

TheAnt1dot3, 2021-02-06:
> First, off. The PWM output is also use to drive the LED on the Teensy so you can disconnect your laser while testing this. The brightness of the LED is proportional to the S-parameter setting in [...]

No question is dumb :)  i'm running 1.1f  from the precompiled hex files you provided. nothing fancy on that side.
I will take some pictures of the wiring and upload them here. for the laser, i'm using one of those chinese '15 Watt' lasers, they have 3 pin connectors for 12v and pwm. this is the one i bought : https://www.wish.com/product/5e6472f81c32cb114b119e91?mi=3&share=web

Photos will follow soon :) 

thanks for your help

TheAnt1dot3, 2021-02-06:
Here are the pictures as requested. 12v input has also the positive and negative (black & red) to the laser connected and PWM to the PWM port on the board. 
https://ibb.co/GJg0W2L
https://ibb.co/92hG3wP

For sending the gcode i tried 2 programs both with the same problem. i tried lightburn and i tried cncjs.
I did note that, when enabled and disabling the laser as expected, the orange LED on the teensy is lit. 
when i move my x-axis to the left (and the laser auto-enables) there is no orange LED lit on the teensy even though the laser is burning at full speed...

phil-barrett, 2021-02-06:
> when i move my x-axis to the left (and the laser auto-enables) there is no orange LED lit on the teensy even though the laser is burning at full speed...

OK, that means grblHAL and the Teensy aren't the ones turning on the laser. Do you have a multimeter? 

Looking at the pictures, it is hard for me to tell how you have connected the laser to the breakout board. I see 7 wires and think this is the arrangement from left to right:
Flood - Black
V+ - Red and Green (2 wires)
G - Blue
PWM - Green
G - Black
+12V - Dark Grey (not sure color)

Where do each of those wires go?  And, I see a Yellow on the Laser connector, where is that connected?

Did the laser come with any hook up instructions? The web site didn't give much info.

TheAnt1dot3, 2021-02-06:
Unfortunatly no hook-up instructions...
Below some details on the wiring
Fld - Black    gnd to relay for controlling air
V+ - Red and Green (2 wires)   5v positive and signal wire to relay for controlling air
G - Blue   to black wire on laser - gnd
PWM - Green to yellow wire on laser - pwm
G - Black   gnd from external 12v power supply
+12V - Dark Grey (not sure color)   combination, 12 v input from power supply and also to red wire of laser for 12v positive

I do have a multimeter. Will post measurements from pwm tomorrow as it´s already past 11 pm here.

Thanks for thinking along

phil-barrett, 2021-02-06:
OK, talk to you in the morning.  

A couple of things to try.  When the laser is firing and it shouldn't be, what is the voltage at the PWM output pin? Also, try disconnecting the relay for air (just the + wire).

TheAnt1dot3, 2021-02-07:
Sorry Phil, it seems the issue is not related to your fantastic board :) i have powered the laser directly to a 12v supply without pwm signal and i can measure the full 12v on the board of the laser.  If i disconnect the 3 wire cable i can only measure the positive and gnd on the cable.. it seems the faulty pwm comes from the laser controller board.. now i need to try and figure out where.. thanks for your help

phil-barrett, 2021-02-07:
I am happy to help with debugging this - if nothing else as someone to bounce ideas off of.  It helps me to provide better service and future products to my customers so I see a lot of benefit from it.

TheAnt1dot3, 2021-02-07:
That´s awesome Phil ! Thanks for that.
I just found out that, if i short pwm to ground it turns off.. 
Could it be, because i had the 12v input connected to the grbhal board input, where i also hooked up the 12v positive to the laser , and then i hooked the pwm and 12v negative to the spindle PWM ports ? Maybe i should try to connect the gnd  from the power supply directly to both laser and the spindle gnd port on the grbhal or wouldn´t that make a difference ?

TheAnt1dot3, 2021-02-18:
Hi Phil,

Because this issue was driving me crazy, i bought a brand new laser (completely different one [link ](https://www.amazon.de/-/en/gp/product/B087T7BB24/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) ) but it is still not acting as it should. Got ground from separate 12V power supply hooked up to both the laser and the grblhal board,  PWM from grblhal to the pwm input for the laser and 12V positive from supply to the laser. as soon as i turn on the power supply, the Laser is getting fired and there is no way to turn it off. i measure about 4V on the pwm port even if the pwm should be disabled through commands (no LED lit on the controller) would it be worth a shot to try a pulldown resistor or something ?

phil-barrett, 2021-02-18:
Hmmm, if the LED on the Teensy is off, the voltage at the PWM terminal should be 0.  Try disconnecting it and see what it reads.  What does the enable terminal read when the LED is not lit.

Looking at that diode laser, it is possible it wants 12V PWM - the board provides 5V.  Not sure why it would read 4V, though.

TheAnt1dot3, 2021-02-18:
i've managed to get it working.. hurray :) 
so, how i hooked it up was, the external 12v supply for the laser was wired directly to the laser and the ground was also hooked up to the grbl board. i connected the ground to the 12v input negative on the board. however, when i measured between that ground and the pwm pin, i kept getting about 3.6 volts no matter the pwm width.
if i measured between the pwn and one of the other ground pins, it was measuring as expected.
once i hooked the ground from the power supply to one of the other ground pins, it started working fine.

i'm guessing there is some separation between the 12v input ground and the rest of the grounds ?

TheAnt1dot3, 2021-02-18:
i will give the other laser another try tonight to see if it works..
and hide the results from my wife... otherwise i've wasted $200 lol

thanks again for your help Phil

phil-barrett, 2021-02-18:
Yeah, getting grounds sorted out can be a little tricky.  I like to think I'm pretty up on that stuff but still get surprised with floating values.

One can never have too many lasers...

closed 2025-03-05 by phil-barrett