phil-barrett/grblHAL-teensy-4.x #8: Component spacing

elfren, 2021-02-06:
While working on an enclosure design for the board I noticed the screw terminals spacing is not consistent.  Some are more than 3.5mm apart and some are less.  Also the upper left hole doesn't have enough space around it for an M3 SHCS.  

Is the .brd file current or is this something which has been fixed in a later version?

phil-barrett, 2021-02-06:
The spacings aren't perfect. On the top row, the rightmost 8 2-pin screw terminals spacing varies between 6.987 and 7.03 mm. The bottom row varies between 6.95 and 7.05 mm. These are all well within the tolerances of the pins for the locking screw terminals. And, even a full row of screw terminals locked together is loose when inserted.  While not intentional, in future boards I will be deliberately varying pin spacings for connectors to tighten them up to make them stay in the holes when inserted. This makes soldering easier since they won't fall out when you turn the board over.

Yes, I am aware of the tight screw hole. It is still in V2.09. In future boards I will take more care of that.  For what it's worth. The 4-40 screws I use are fine.  That is no comfort to my European customers, though. I am now validating my designs with metric harware (as well as imperial).

elfren, 2021-02-06:
I just imported the .brd file into Fusion 260, then imported the M3 SHCS from McMaster-Carr.
If I understand you correctly, you're going to change the pin spacing in the connectors components to be less than or greater than 3.5mm, but the center to center spacing between components will be the spec distance.  That sounds like a good plan.

Thanks.

closed 2021-02-06 by elfren

phil-barrett, 2021-02-06:
Yes.  Definitely for 4 pin connectors.  Still deciding for 2 pin ones. I've made a couple of test boards and it seems ok. This is especially nice for pin headers that not only fall out but are hard to get perfectly straight.  I plan on making fully assembled boards available and am worrying about DfM (Designing for Manufacturability). I want all the headers and terminals perfectly straight. One thing that really bothers me is when I buy a board and find wildly misaligned connectors.  Says the manufacturer doesn't care.