phil-barrett/grblHAL-teensy-4.x #9: Compiling help

Cncninja, 2021-02-10:
I've been fighting with this for a few weeks, and wonder if you could just make a precompiled version of the 3axis for lasers. I need the buffer changed to 1024 so I can move faster I have a 60W laser so I can move a bit faster than the LED lasers. Ive tried everything to compile, Ive used your guide but ran into the same thing as others with different/missing folders. I tried compiling using the most current Grblhal build and instructions and get an error ( no IOport.h file) but can't find where or what is even calling for it.  I just need a 3 axis with homing switches (No Z tho) and the buffer set to 1024 and the 30k step limit removed.

Thanks

Cncninja, 2021-02-10:
[image]

phil-barrett, 2021-02-10:
Here you go.  [3, 4 and 5 axis with planning buffer set to 1024](https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/grblHAL_pb1024.zip). 30K limit does not exist in grblHAL.

Cncninja, 2021-02-10:
Thanks, I really appreciate it, any idea about the ioports.h error i would really like to understand why its not working for me, also if grblhal does not have step rate limits why is it still defined in the config file

// Sets the maximum step rate allowed to be written as a Grbl setting. This option enables an error
// check in the settings module to prevent settings values that will exceed this limitation. The maximum
// step rate is strictly limited by the CPU speed and will change if something other than an AVR running
// at 16MHz is used.
// NOTE: For now disabled, will enable if flash space permits.
#define MAX_STEP_RATE_HZ 30000 // Hz

phil-barrett, 2021-02-10:
OK, forgot about that.  Never had that enabled for any of my builds.

I'm not sure why you are seeing the ioports error.  could be a mix and match thing. I built with the master branch and keep it pure.  If you download the test over the master branch (or the other way around) you might wind up with left over files. The arduino IDE by default will try to compile everything.

Cncninja, 2021-02-11:
Worked like a charm !!! the laser is up and running.

phil-barrett, 2025-03-05:
To bring this forward, the grblHAL Web Builder will allow people to build their own firmware without needing to compile.

closed 2025-03-05 by phil-barrett