phil-barrett/grblHAL-teensy-4.x #10: Npn or pnp 3 wire inductive sensors

raenji-sk, 2021-02-22:
Hi, does the breakout board support the inductive sensors? I am currently running some 6-30V PNP NC ones with my current arduino setup and would like to continue using them as I find them very reliable. I see you have some boards in stock on tindie and want to get one, but this is something I have not found in your documentation and I am not sure they are supported. Can they be powered by an external PSU and plugged in your board. You say the inputs are optoisolated, so I think it should be fine, can you please confirm this? Also are all 5 endstop inputs isolated?

phil-barrett, 2021-02-22:
I am using inductive sensors on one mill but am running them at 5V,  They were listed as 6-24V.

All 5 limits go through optoisolators but share a common ground with the processor.  If your sensors are open collector then they should work. If they are "Push-Pull" or sourcing (as opposed to sinking), then not as it will feed high voltage back into the power supply.

I am putting the finishing touches on a Pro version of the breakout board that has truly isolated inputs and spindle. Optos runs off 12V. Pluggable screw terminals,  Maybe ships in a month.

raenji-sk, 2021-02-22:
Alright, thanks for the reply. I'll have to double check on that, in worst case I'll get some new sensors, mine do not work at 5V for sure, I run them at 12V at the moment. I did make the purchase in the meantime so I'll have to figure it now anyway 😉. The grbl HAL and  your board has been on my mind for a long time. I really like it does not use the step sticks and rather than that it has outputs for external drivers among other things. I'll be checking for updates regarding the pro version for sure and eventually get it later on after getting a bit familiar with this one and seeing how it performs compared to the Arduino GRBL. I guess you can never have enough hardware handy for any possible new projects.

phil-barrett, 2021-02-22:
>
> I guess you can never have enough hardware

Truer words were never spoken!

I do lose a lot of sales to step stick boards.  I just can't imagine using
them to drive a spindle through anything.

That said, I will probably release a board supporting them - mainly for
small laser machines.  The new TMC5160 is a very capable stepper driver and
can handle a lot higher current.  Too bad those pin headers do not.

raenji-sk, 2021-02-23:
> Too bad those pin headers do not.

That is a very good point.

raenji-sk, 2021-03-25:
Got the board up and running. Inductive endstops work without issues. Just had to solder some 1k8 resistors between signal and ground so they get pulled to zero. Will do some test cuts tomorrow.

raenji-sk, 2021-03-26:
Everything works great using bCNC as GCode sender. Closing Issue.

closed 2021-03-26 by raenji-sk

sogeniusio, 2023-02-15:
> Got the board up and running. Inductive endstops work without issues. Just had to solder some 1k8 resistors between signal and ground so they get pulled to zero. Will do some test cuts tomorrow.

Is this the official method for getting an inductive probe to work with the grblHAL probe? How reliable is soldering those 1k8 resistors?