phil-barrett/grblHAL-teensy-4.x #14: Aux Relay Outputs?

mlewus, 2021-05-21:
Does anyone know if there is any standard mapping for the use of the Aux 0, 1, & 2 relay outputs in GRBL on the breakout board?

raenji-sk, 2021-06-07:
http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65
I use theese to control one of the aux pins. Had to modify the postprocessor for fusion 360 so that it triggers it. I use it to turn on/off the vacuum independently of the spindle.

ericspooner, 2022-07-12:
For anyone finding this, I found based on the above:

M64 P0 = Aux 0 
M65 P0 =Aux 0 off

M64 P1 = Aux 1 
M65 P1 =Aux 1 off

M64 P2 = Aux 2
M65 P2 =Aux 2 off

I'm guessing M62PX and M63 PX Spindle On/Off and Vac On/Off, but I didn't get +5V out of them, though I haven't spent much time with the board yet.

phil-barrett, 2022-07-12:
Spindle and (and Vac) are controlled by M3 (on, forward), M4 (on, reverse) and M5 (off).

FWIW, M7 turns Mist on, M8 turns Flood on and M9 turns them both off.

The LinuxCNC GCode manual describes the M62-M65 behavior. [Relevant section](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65).

ericspooner, 2022-07-12:
Thanks for the response! I got my board this weekend, and I did notice those turned on, but wasn't sure if you could separate them to use the M62's or not. Looking forward to more testing with the board!

cuishuang413, 2024-01-19:
> For anyone finding this, I found based on the above:
>
> M64 P0 = Aux 0 M65 P0 =Aux 0 off
>
> M64 P1 = Aux 1 M65 P1 =Aux 1 off
>
> M64 P2 = Aux 2 M65 P2 =Aux 2 off
>
> I'm guessing M62PX and M63 PX Spindle [...]

Hello everyone, I tried M64 and M65, it doesn't look like it's doing anything. I'm using a custom grblHAL firmware, is there some config I need to define to enable it?

Thanks,

phil-barrett, 2024-01-20:
Where are you testing the voltage?  At the screw terminals or the pin headers?

The screw terminals are open collector outputs so will not show a voltage.  These essentially act like a switch.  [This](https://www.grbl.org/single-post/using-relays-with-grblhal-and-the-t41u5xbb) describes it in more detail. By the way, you need to jumper the relay voltage selection header, as well.

For more info on using the open collector screw terminals, [check here](https://www.grbl.org/single-post/ttl-ssr-on-the-screw-terminals).

cuishuang413, 2024-01-20:
I was testing them on the pin headers. And it turns out that I didn't have the correct board selected. Thanks anyways!

Perlovka, 2025-01-13:
>The screw terminals are open collector outputs so will not show a voltage. These essentially act like a switch.
>By the way, you need to jumper the relay voltage selection header

These two sentences are mutually exclusive, aren't they?
I've set jumper to 12v, issued M3 S6000 and spindle relay is showing 12v on screw terminals.
So this can not act like a simple switch, isn't it?

On my inverter there is X1 output 12v that should be pulled to GND to run spindle. But there is no option on the board for such a function as far as I understand from documentation.

phil-barrett, 2025-01-13:
It does act like a switch but not between the two relay coil terminals.  The relay coil drivers are open collector outputs.  The +V pin is connected to the relay voltage selected by the 3 pin header. The signal output is connected to the collector of a darlington transistor.  When the transistor is turned on (signal output high) it "connects" the collector pin (the one not labeled +V) to ground.  The relay coil needs to be connected the +V pin and that pin.  Current will flow through the relay coil when the output signal turns on the transistor. Hopefully, this diagram will help.

[image]

stevempotter, 2025-01-14:
When designing your boards, why didn't you use an FET to switch relay coils rather than a Darlington pair of transistors?

phil-barrett, 2025-01-14:
A MOSFET has advantages over a bipolar darlington pair but there are additional components needed for either - resistors to limit base (or gate) current, a pull down to prevent glitches on startup and a diode to protect the transistor(s) from back EMF. The ULN2003 has all that in a single package for 7 outputs. A MOSFET could handle higher current but for the purpose of driving relay (and perhaps solenoid) coils, the ULN2003 is a reasonable and cost effective choice.

stevempotter, 2025-01-14:
Thanks for that great comparison of MOSFETS vs Darlingtons, and explanation of your design choices!

phil-barrett, 2025-01-14:
No problem.  One other thing about bipolar transistors is they are pretty rugged which I think is important for this application. It doesn't hurt that they are fairly cheap.

BTW, there is a mosfet version of the ULN2003 that should drop in without modification. However it is several times more expensive (probably due to significantly lower production volume). Given this is a low cost product, I am hesitant to add cost without a very clear benefit.

closed 2025-03-05 by phil-barrett