phil-barrett/grblHAL-teensy-4.x #17: $help command does not return the list of settings

krupis, 2021-06-15:
Hello. I have previously used the grblhal firmware on the esp32 device. I was able to write $help in terminal and it would respond with all the possible settings. For example $help limits would return:
[image]

Is there such possibility with teensy 4.1 grblhal? I have flashed the teensy with 3axis compat firmware but $help does not return setting list.

phil-barrett, 2021-06-15:
Switch to the newer builds I mentioned - you will see this:
[image]

krupis, 2021-06-17:
Hello. Thanks for the response. I have flashed the device with the ethernet configuration. Typing $Help in terminal still does not show the menu that you have provided. Can you explain how to invoke this help arguments menu? Also, could you please send me which documentation I should refer to when using grblhal?  Simply googling "grblhal command list" returns this one : 
https://github.com/terjeio/grblHAL/blob/master/doc/markdown/settings.md

But if I understand correctly, this is not the latest documentation, please can you confirm which one should I use?

phil-barrett, 2021-06-17:
What build of the firmware did you flash? Which sender are you using?

krupis, 2021-06-17:
Im using the first build ( 3 axis with enc) and using bCNC

phil-barrett, 2021-06-17:
Ah, you need to move to a later build (Feb 2021 trimeframe or later).  You can use the ethernet builds I put up.  USB works fine with them. [Look here](https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/prebuilt.md) and scroll down to the ethernet build section

krupis, 2021-06-17:
Okay tommorow I wil try to flash download configuration from your link and
flash with 3 axis quad enc ethernet build

krupis, 2021-06-18:
Hello. I have downloaded the ethernet_configurations folder from the link you have provided : https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/ethernet_configurations.zip . I have flashed the teensy with the 3 axis with quad enc. The behaviour is exactly the same, I do not get any information display when typing $help. Please can you assist me further to find a cause to this? 

After connecting and typing $help:
'''
ok

ok
$#
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G59.6:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
ok
$help
error:5
$#
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G59.6:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
ok
'''

I can see that the HEX file I have used for flashing was last modified January 29. Would this be the correct HEX?

phil-barrett, 2021-06-18:
What does $i report?

krupis, 2021-06-18:
```
$i
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FLASH,ES,ETH,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[IP:0.0.0.0]
[PLUGIN:ENCODER v0.01]
ok
b'$G\n'
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
ok
```

phil-barrett, 2021-06-18:
Ah, just checked the binary.  I'm sorry, needs to be newer. Use this.  I will update the zip file in a bit.
[3 axis eth.zip](https://github.com/phil-barrett/grblHAL-teensy-4.x/files/6674612/3.axis.eth.zip)

krupis, 2021-06-18:
Yes! Works as expected now. Surprised no one has noticed that before :o

closed 2025-03-05 by phil-barrett