phil-barrett/grblHAL-teensy-4.x #18: RJ45 MagJack

karoria, 2021-07-29:
Hi @phil-barrett 
I am curious about what "magjack" really is, and is it compulsory requirement for ethernet connection on teensy4.1? I am designing my custom PCB for grblHAL. Can I use regular shielded RJ45 pcb connectors?
I plan to use RJ45 mounted on PCB with tracks instead of using that popular kit with flat cable.

phil-barrett, 2021-07-29:
Ethernet requires a very specific set of transformers, "magnetics", to work.  In the early days of twisted pair ethernet they were separate but most ethernet products use the version that is integrated into the RJ45 jack - thus MagJack - short for magnetic jack.

You can use any 100BT magjack but you will need to make the correct connections - they are all different. Unfortunately, the datasheets tend to not give much detail about how to connect. Probably your best bet is to use the PJRC magjack kit as the basis for your design.

Here are 2 docs that can help you dig deeper. Unfortunately, the info is buried in the middle of them
https://www.ti.com/lit/an/snla079d/snla079d.pdf?ts=1591489614886&ref_url=https://www.google.com/
https://www.analog.com/media/en/technical-documentation/application-notes/EE-269.pdf

karoria, 2021-07-29:
@phil-barrett, that is completely new to me. Till now I was understanding that all RJ45 are same and doesn't have any circuitary inside it. Thanks for pointing me to right direction and reading material. I will manage from it. Thanks again for your time.
Regards,
Ravi

closed 2021-08-11 by phil-barrett