phil-barrett/grblHAL-teensy-4.x #19: Compiling with Backlash compensation Enabled

humneraudio, 2021-08-11:
Hi,

I've been using the breakout board with Ethernet and bCNC - its running very well. I currently run it on Linux with a dedicated NIC to the teensy breakout board. I used the prebuilt firmware "3 axis eth quad enc.hex"

If I trigger $$ it doesn't return any $160/$161/$162 values, so I assume backlash compensation isn't enabled in this firmware build. Is that correct?

I want to try backlash compensation - however I feel its a little out of my depth in compiling a custom firmware that matches the 3 axis eth quad enc.hex firmware and includes backlash compensation. As everything already running so well, I'm concerned there maybe some specific options enabled which I might miss if I were to compile something myself.

I need some guidance, or if someone can build me the hex that matches the "3 axis eth quad enc.hex" + backlash compensation, that would be great.

phil-barrett, 2021-08-11:
You are lucky!  I am heading to airport in an hour and will be gone for a month.  Here is a hex that supports backlash. There seems to be a problem with the encoder plugin and I don't have time to track it down.  Hope it is not vital to you.

Also, could you post details of how you got ethernet working with bCNC on the RasPi?

[3X_eth_backlash.zip](https://github.com/phil-barrett/grblHAL-teensy-4.x/files/6969828/3X_eth_backlash.zip)

humneraudio, 2021-08-11:
Hi Phil,

Thanks! I will give this a go later today.

Regarding Ethernet on bCNC - 

My CNC is a fixed gantry 3 axis machine - the only additional auxirlary components wired in are are NC Limit switches and NO eStop, RUN and HOLD buttons. I have homing enabled with soft limits.

I'm running latest Lunbuntu with the latest bCNC install - no special config going on to make this work. The PC i'm running has dual NIC, so I have a single network cable connected between one NIC and the teensy, the other NIC is for my normal network.

I setup your teensy board as per your instructions and then did the following for Ethernet to set a static IP for that single NIC/network...

```
$301=0
$302=192.168.60.200
$303=192.168.60.1
$304=255.255.255.0
```

Then in bCNC I use the following in the Serial Port Field

`socket://192.168.60.200:23`

And it connects like normal. While its connected its rock solid, I leave bCNC running all the time even when i leave my workshop(however I power down the stepper drivers). Its remained connected like this for weeks without issues.

I noticed that your default $10=511 value seems important for bCNC to run correctly and report correct coords/status.

The only "major" issue I've noticed is if I close bCNC, or close the connection through bCNC, most of the time the ethernet side of the teensy seems to crash and I can't reconnect it until I power cycle the teensy OR I engage a serial connection via USB and call $$ - that seems to "wake it up" so it speak. This issue doesn't effect me so much as I leave bCNC open and connected all the time - but it should be noted nonetheless.

If you have any other questions or testing I can do, let me know, happy to help.

Cheers

Rob

humneraudio, 2021-08-16:
Hi Phil,

Thanks again for producing the hex file for me, especially considering you were rushing to get onto a plane!

The hex loaded up fine, mostly all good, but It looks like gantry squaring is enabled in this build and I don't think there is a $ config to turn it off. I only have a single Y limit switch and don't use the squaring function. 

I've done a work around for now - I connected the Y limit output to the A limit output so they sink to ground via the same limit switch. This tricks grblHAL into thinking there are 2 limit switches and it homes OK now.

And the backlash compensation works perfectly - this was one of the reasons I had for moving to grblHAL. I already optimised my backlash as much as I could mechanically and software compensation makes sense to me to get rid of the rest.

I use delrin acme t8 nuts and got it down to 0.07mm on X, 0.05mm on the Y and 0.08mm on the Z. I entered these into the backlash compensation and the backlash is now gone! So happy this worked straight out of the box.
I've run a few cuts already on it and it's working well. 

Cheers

Rob

phil-barrett, 2021-08-16:
Sorry about that. When I return home I will make new one without ganged Y.
Please remind me around Sept 6.

phil-barrett, 2025-03-05:
To bring this question current, with the grblHAL Web Builder, you can now build backlash enabled firmware.