phil-barrett/grblHAL-teensy-4.x #21: Auto square gantry. Which hex?

Shkolik, 2021-09-06:
Hello. 
Could you please help me with right hex for ethernet board. My cnc has 2 motors on Y so I want to use auto square Y when homing. On a board itself I cut 2 connections and bridge other 2 based on manual for ganged Y. Tried to use "3 axis gangY eth quad enc.hex" but homing not working in this configuration at all. Now I flashed "4 axis gangY eth quad enc.hex" and homing working but looks like not squaring - homing sequence ends on Y limit switch, "A" looks like ignored. 

Should I change something in HW or flash another hex?  

Thank you!

phil-barrett, 2021-09-06:
Sorry for the troubles. The hardware config was intended for use before grblHAL supported dual motor axis. It duplicates the Y signals on B (or is it A?, I'm on the road so can't look at docs). As such, it does not support squaring.  To use squaring,  you will need to undo the 2 changes.  I'm back home later today (yay! 26 days on the road) and can give you more help then.

I will update the manual to note this. Sorry for the waisted effort.

Shkolik, 2021-09-06:
Phil, no reason to be sorry! You did wonderful work with this board! I'm just asking for help, because not really familiar with cnc :) will be waiting for your response with patience.

Than you!

phil-barrett, 2021-09-07:
Got home.  I configured my_machine.h for 3 axis, ganged Y, auto-square and ethernet. If you don't use ethernet, you can just ignore it - no penalty. I have attached the generated .hex file.  You will need undo to the solder configuration in the "Link A Axis to Y" box on the bottom of the board.  Use solder wick to removed the Y bridges and bridge the A ones. 

[grblHAL_T41E3XGYAS.zip](https://github.com/phil-barrett/grblHAL-teensy-4.x/files/7118359/grblHAL_T41E3XGYAS.zip)

Here's my_machine.h for reference. 
```
/*
  my_machine.h - configuration for IMXRT1062 processor (on Teensy 4.x board)

  Part of grblHAL

  Copyright (c) 2020 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// BOARD_T40X101 and BOARD_T41U5XBB by Phil Barrett: https://github.com/phil-barrett/grblHAL-teensy-4.x

// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used
//#define BOARD_T40X101
#define BOARD_T41U5XBB
//#define BOARD_T41U5XBB_SS // For a modified T41U5XBB board, allows spindle sync to be enabled.
//#define BOARD_T41BB5X_PRO
//#define BOARD_CNC_BOOSTERPACK
//#define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!

// Configuration
// Uncomment to enable, for some a value > 1 may be assigned, if so the default value is shown.

/*
              Plugin: | ETHERNET¹ | SDCARD¹ | KEYPAD | EEPROM | N_AXIS |
----------------------|-----------|---------|--------|--------|--------|
BOARD_T40X101         | no        | no      | yes    | yes³   | max 4  |
BOARD_T41U5XBB        | yes       | yes     | yes    | yes³   | max 5  |
BOARD_T41BB5X_PRO     | yes       | yes     | yes    | yes    | max 5  |
BOARD_CNC_BOOSTERPACK | yes²      | yes     | yes    | yes    | max 3  |

¹ Teensy 4.1 only
² External magjack.
³ EEPROM is optional and must be added to the board.

N_AXIS has a default value of 3, edit grbl\config.h to increase.

*/

#define USB_SERIAL_CDC      2 // 1 for Arduino class library and 2 for PJRC C library. Comment out to use UART communication.
//#define USB_SERIAL_WAIT     1 // Wait for USB connection before starting grblHAL.
//#define BLUETOOTH_ENABLE    1 // Set to 1 for HC-05 module. Requires Bluetooth plugin.
//#define SPINDLE_HUANYANG    1 // Set to 1 or 2 for Huanyang VFD spindle. Requires spindle plugin.
//#define QEI_ENABLE          1 // Enable quadrature encoder interfaces. Max value is 1. Requires encoder plugin.
#define ETHERNET_ENABLE     1 // Ethernet streaming. Requires networking plugin.
//#define SDCARD_ENABLE       1 // Run gcode programs from SD card, requires sdcard plugin.
//#define KEYPAD_ENABLE       1 // I2C keypad for jogging etc., requires keypad plugin.
//#define PLASMA_ENABLE       1 // Plasma/THC plugin. To be completed.
//#define MCP3221_ENABLE      1 // Enable analog input via MCP3221 ADC.
//#define PPI_ENABLE          1 // Laser PPI plugin. To be completed.
//#define ODOMETER_ENABLE     1 // Odometer plugin.
//#define OPENPNP_ENABLE      1 // OpenPNP plugin. To be completed.
//#define FANS_ENABLE         1 // Enable fan control via M106/M107. Requires fans plugin.
//#define EEPROM_ENABLE       1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 for larger sizes. Requires eeprom plugin.
//#define EEPROM_IS_FRAM      1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.
//#define SPINDLE_SYNC_ENABLE 1 // Enable spindle sync support (G33, G76). !! NOTE: Alpha quality - enable only for test or verification.
                                // Currently only available for BOARD_T41BB5X_PRO and BOARD_T41U5XBB_SS.
//#define ESTOP_ENABLE        0 // When enabled only real-time report requests will be executed when the reset pin is asserted.
                                // Note: if left commented out the default setting is determined from COMPATIBILITY_LEVEL.

// If the selected board map supports more than three motors ganging and/or auto-squaring
// of axes can be enabled here.
//#define X_GANGED            1
//#define X_AUTO_SQUARE       1
#define Y_GANGED            1
#define Y_AUTO_SQUARE       1
//#define Z_GANGED            1
//#define Z_AUTO_SQUARE       1
// For ganged axes the limit switch input (if available) can be configured to act as a max travel limit switch.
// NOTE: If board map already has max limit inputs defined this configuration will be ignored.
//#define X_GANGED_LIM_MAX    1
//#define Y_GANGED_LIM_MAX    1
//#define Z_GANGED_LIM_MAX    1
//

#if ETHERNET_ENABLE > 0
#define TELNET_ENABLE           1 // Telnet daemon - requires Ethernet streaming enabled.
#define FTP_ENABLE              1 // Ftp daemon - requires SD card enabled.
#define WEBSOCKET_ENABLE        1 // Websocket daemon - requires Ethernet streaming enabled.
#define NETWORK_HOSTNAME        "GRBL"
#define NETWORK_IPMODE          1 // 0 = static, 1 = DHCP, 2 = AutoIP
#define NETWORK_IP              "192.168.5.1"
#define NETWORK_GATEWAY         "192.168.5.1"
#define NETWORK_MASK            "255.255.255.0"
#define NETWORK_TELNET_PORT     23
#define NETWORK_WEBSOCKET_PORT  80
#define NETWORK_HTTP_PORT       80
#endif

```

Shkolik, 2021-09-07:
Thank you! I already build my own, but for some reason build ends with error when I enable encoder (#define QEI_ENABLE 1). Without this param everything works. Have no idea - maybe it's a bug or maybe it's me doing something wrong :) Could you try to build the same configuration (3 axis, gangY + auto square Y+ network) + encoder? 

Thanks for your help! It's invaluable!

phil-barrett, 2021-09-07:
Yeah, I am getting a bunch of errors.  Maybe Terje did not update the encoder source code.  I put some effort into it tomorrow - fading fast from  27 hrs in transit.

phil-barrett, 2021-09-08:
Here is a link to the issue - it is being fixed and [here is a work around](https://github.com/grblHAL/iMXRT1062/issues/13) for you if you don't want to wait.

Shkolik, 2021-09-08:
Will try today. Thank you!

Shkolik, 2021-09-10:
> Here is a link to the issue - it is being fixed and [here is a work around](https://github.com/grblHAL/iMXRT1062/issues/13) for you if you don't want to wait.

Workaround kinda works but only with 4 days old code. Now repository for teensy produces even more errors. Looks like Terje  doing refactoring.

closed 2021-09-10 by Shkolik

phil-barrett, 2021-09-10:
Here's a build with the workaround that Terje suggested.  3 Axis, Y ganged to A, encoder, ethernet.
[grblHAL_T41E3XGYAETHQEI.zip](https://github.com/phil-barrett/grblHAL-teensy-4.x/files/7145227/grblHAL_T41E3XGYAETHQEI.zip)

Shkolik, 2021-09-10:
Thank you, I was able to build it myself.