phil-barrett/grblHAL-teensy-4.x #24: Using an inductive proximity sensor as a limit switch

molove, 2021-11-20:
Hello

Not an issue per se, but a request for help.
I'm a bit of a noob when it comes to electonics/arduinos etc. 

I have the breakout board which I've soldered up and am in the process of building an Openbuilds Sphinx type cnc, I would like to make use of grblHAL's auto square function and would like to use proximity sensors instead of limit switches. 

I'm not entirelt sure how they work but I assume that normally the limit switches just short the limit input to ground.
I have watched this youtube vid https://www.youtube.com/watch?v=F7BjZElStfQ so know how get the output of the inductive sensor to 5v. If I take the power for the inductive sensor from the same supply as the 12v input and using a suitable pulldown resistor on the sensor output, do I just connect the sensor output to the limit input, and when the sensor output goes low it triggers the limit input?

I don't want to damage the board or teensy by sticking 5v onto an input that might not be expecting it. 

Great job with the board btw, really quick delivery to the UK.

TIA

Piers

phil-barrett, 2021-11-20:
Thanks for linking to that video.  Worth watching.  The inputs for the limit switches are looking for 5V, anything below 5.25V will be good. The 2K he uses in the video is a bit low, try 2200 or 2700 to get the voltage to a little below 5V.  You can go as low as 4.25V without any problems.

Note that the actual resistor value will be dependent on the pull-up resistor in the sensor. They are all going to be different so you will have to experiment with yours like in the video. 

To answer your direct question, yes, you would put a resistor between input and ground along with the sensor input and common ground.  If you use 22 Ga wires, you should be able to inset both the resistor lead and senor wire into the screw terminals.  I recommend using a ferrule to keep the wires together. 20 Ga ferrules are a bit tight but can be made to fit the screw terminals.

phil-barrett, 2021-11-20:
Here is the github discussion on this.
https://github.com/phil-barrett/grblHAL-teensy-4.x/issues/24

I may do a blog on this topic.

molove, 2021-11-20:
Thanks for the swift reply, and confirming I had understood correctly.

 I will adjust the resistor to match my sensor so it's outputting less than 5.25V. And yes I'm very fond of a ferrule and try to use them for all my projects.

I'm sure there will be more questions that I have as I set up my machine, that won't be "issues" as such. Is there a proper forum to ask them on, or should I just ask here?

phil-barrett, 2021-11-20:
You are welcome and, yes, you got it right.

I enabled discussions for this repository. So, [you can ask them here](https://github.com/phil-barrett/grblHAL-teensy-4.x/discussions).

stevempotter, 2021-11-21:
@molove, see my post on here on the Discussion forum about using Hall Effect switches as endstops.

molove, 2021-11-21:
@stevempotter I'm not sure where I'm meant to be looking for your post, can you let me have a link?

phil-barrett, 2021-11-21:
> @stevempotter I'm not sure where I'm meant to be looking for your post, can you let me have a link?

It's [here, scroll down a bit.](https://github.com/phil-barrett/grblHAL-teensy-4.x/discussions/25)

humneraudio, 2023-05-22:
With regards to NPN proximity sensors, which is best between NC or NO?

phil-barrett, 2023-05-22:
It is a bit complex. If you are only going to use one per axis, then either NO or NC. However,  if you will be using 2 per axis, then NO is better. This assumes that you are using sensors with open collector outputs.

humneraudio, 2023-05-22:
no worries, thanks. it will be a single sensor per limit input.

humneraudio, 2023-08-21:
I thought I would just use this to continue with this discussion. I setup the NPN NC proximity sensors and used resistors to get the same 5 v output for the BOB. I used a 12v SMPS for the proximity sensors.

I prefer to have the microusb powering the BOB via my raspberry PI - however I found the proximity sensors don't work in this config. I assume its because both grounds are floating? 

If I power the BOB/teensy via a 5v buck converter on the 12v SMPS - it all works ok as it appears the GND is linked.

However I still have a preference for the raspberry PI to power the teensy.

Does that mean I should connect the GND from the 12v SMPS to the GND of the BOB/teensy if its connected to the Raspberry PI to make it a common GND point?

Is that safe?

phil-barrett, 2023-08-21:
Yes, it is safe to connect the two grounds.  They need a common ground otherwise the sensor voltage is not referenced to the ground of the breakout board (which is the Pi's ground, too).

humneraudio, 2023-08-25:
thanks phil. I had a chance to play around with it today, however joining the grounds didn't have any effect or change while the breakout board was connected to the Raspberry PI. 

I think I need to take a step back and double check my wiring before i continue.

humneraudio, 2023-08-26:
long story short, it was a wiring fault. And in the process, I also found that I had tied the ground of the breakout board to chassis ground via a DIN cable I made years ago for the halt/start/hold buttons. I had tied the DIN cable shield to the ground connection within the cable itself. No idea why I did that, but have at least solved some mysteries.

closed 2025-03-05 by phil-barrett