phil-barrett/grblHAL-teensy-4.x #44: Ethernet Pairs used

gxdeange, 2022-01-27:
Hi phil-barrett,

Just a quick question regarding the wire pairs used in the schematic for ethernet connectivity.

I noticed you are using pairs 2 (Green) & 4 (Brown) but I suspect it should be pairs 2 (Green) & 3 (Orange).

This is based on T-568a.

Cheers
George

phil-barrett, 2022-01-28:
> Hi phil-barrett,
> 
> Just a quick question regarding the wire pairs used in the schematic for ethernet connectivity.
> 
> I noticed you are using pairs 2 (Green) & 4 (Brown) but I suspect it should be pairs 2 (Green) & 3 (Orange).
> 
> This is based on T-568a.
> 
> Cheers
> George
> 

Thanks for the comment.  

To be honest, I am not sure there is a problem.  This is standard networking with the Teensy 4.1.  Many people ( multi hundreds) are using the T41 breakout board with Ethernet.   And a lot more people are using Ethernet with the Teensy 4.1 plus the PJRC Ethernet kit which uses the same wiring as I do.

gxdeange, 2022-01-29:
OK. I just realised that this is not a standard RJ45 jack. 

I also noticed this diagram which I believe is the PJRC Ethernet kit schematic.

[image]

Can you please confirm this is what you have used?

Cheers
George

phil-barrett, 2022-01-29:
That's not the specific magjack I use but yes, it's not an plain RJ45. It has integrated magnetics. There is no standard for magjacks - every one is slightly different so there is no interoperability.

gxdeange, 2022-01-29:
Awesome. I found a MagJack with equivalent parameters but slightly different pin assignments.

Cheers and have a great day
George

closed 2022-01-29 by gxdeange

phil-barrett, 2022-01-30:
By the way, in the manual for the Teensy BOB, I have the part number of the I use in the BOM.

karoria, 2022-01-30:
> Awesome. I found a MagJack with equivalent parameters but slightly different pin assignments.
> 
> Cheers and have a great day
> George

There are magjacks with 2 or 3 types of pin assignments in the market. You can check continuity by multimeter to see which pin diagram is suitable for what you have. Pins connected with coils should show continuity.