phil-barrett/grblHAL-teensy-4.x #45: Vacuum controls trouble shooting

Drew265, 2022-02-02:
The capability of this board has motivated me to incorporate dust collection onto my machine. I added a relay to control a small shop vac and attached the relay to the TTL/SSR Relay pins labeled VAC/DC. The vacuum cycles on and off with the spindle as intended. 

I then attached a switch to the Vac/DC Trigger pins and that is where my trouble begins. The vacuum does not turn on when the switch is toggled. I have done the following to find the problem.

1. Check the switch is working correctly.
2. Checked pin 1 of JP8 for 3V.
3. Checked pin 2 of JP8 to ground for 100K resistance.
4. Checked continuity between pin 2 of JP8 to IC16.

My assumption is that because spindle enable also feeds into IC16 that everything from this point on is working correctly.

Is that assumption fair?
What additional checks should I perform to trouble shoot this issue or is the problem accurately assessed to be within IC16?

phil-barrett, 2022-02-02:
The switch should work. Spindle enable and the VAC/DC Trigger pin header (pin 2) feeds into an OR gate (IC16). Pulling pin 2 to 3.3V (i.e connecting to pin 1) should cause the the output of the OR gate to turn on the VAC Relay output.  If Spindle Enable is doing that, the OR gate is working. You said pin 2 connects to IC16 (top left pin of IC16, when looking at the board with the Teensy to the left) and pin 1 reads 3V (probably closer to 3.3V). That is as it should be.

Are you sure the switch is working and the wires are connected correctly to the switch?  Given what you said above, that seems like the most likely cause. Try carefully shorting the two pins with a small metal screw driver to see if the relay turns on.

Drew265, 2022-02-03:
Thanks for the quick response! Yes, I measured 3.3v not 3 like you suggested.
I placed a jumper across the pins where the switch connects to the board and the relay does not turn on. An interesting observation, while the jumper was installed, I turned on the spindle just to see what would happen and the vacuum did not turn on. I then removed the jumper while the spindle was still on and the vacuum turned on. 
And then the light bulb came on.... I reversed spindle enable in the GRBL settings ($16=1).
I changed $16=0 and the vacuum works correctly with the switch.
Looks like I need to rethink how I have the spindle setup.  I will clean that up and see if I can get everything working together.

Thanks a lot Phil, I would not have gotten this far without your explanation of the circuit.

phil-barrett, 2022-02-03:
Glad to help.  Yes, you will need to have spindle enable not inverted to be able to use the vac switch.  Though, I don't understand why the vac wasn't running when the spindle was off with spindle enable inverted. I will have to set that up an look at it.

closed 2025-03-05 by phil-barrett