phil-barrett/grblHAL-teensy-4.x #71: Connecting T41U5XBB to a servo drive

tonluc123, 2022-12-25:
Hello,
My goal is to run T41U5XBB with a servo drive. So far I run T41U5XBB with steppers and it runs perfect. The servo drives and motors that I have available are Siemens V90 and they accept signals 24v or High-speed 5 V differential pulses. I tried driving directly with Teensy firmware and T41U5XBB but I'm missing pulses. My thought is that since I have not differential pulses it would be a good idea to amplify the 5 signals to 24v since no differential signals are needed at this drives input. My question (request) is this: what is the best way to go ?
High speed optocouplers, high speed transistors ? Any suggestions are welcomed.
Lucas

phil-barrett, 2022-12-25:
Hmmm, could you give me a link to the manual or datasheet for the servo controller?  It might be as simple as increasing pulse width but lets see the servo wants.

tonluc123, 2022-12-26:
Hi Phill, Merry Christmas,
This is the link (page 60) https://cache.industry.siemens.com/dl/files/218/93005218/att_74528/v1/SINAMICS_V90_OPI_en-US_en-US.pdf
I do not think the pulses width is the problem, I have driven it with a PLC and checked with an oscilloscope, at high speeds the width is the same (I'm not able to do this right now and verify it again).
Thanks for your fast response.
Lucas

phil-barrett, 2022-12-26:
Looked through the manual.  Whew! Now that's DOCUMENTATION. 

I wasn't able to see if it actually takes simple step/dir/enable signals but assume that is possible. Maybe there is a separate interfacing document. 

It could be output signal drive from the T41U5XBB - it should handle 10 mA. But if the input optos of V90 expect higher voltage, it could be they have higher resistance that drops the current.  FWIW, I have never seen this problem with the T41U5XBB after shipping over 1000 units.

I would use a simple driver circuit using an NPN transistor or something like a ULN2003 (which has 7 drivers in a single package). You'd need to invert the signals in the Grbl settings. Since the V90 already has optos inside, you don't need to worry about that.  I can work up a circuit drawing if you need.

tonluc123, 2022-12-26:
Phill,
Yes the drive has step/dir/enable signals. The drive has different inputs for 5 and 24vdc (High-speed 5 V differential pulse train input (RS485)Maximum frequency: 1MHz=1-2 steps & 26-27 dir) and (24 V single end pulse train input Maximum frequency: 200 kHz 36-37 steps & 38-39 dir). Of course this is not a T41U5XBB problem, this board was designed for steppers and their drives and as I said it works fine with them. ULN2003 was my first choice regarding voltage and speed but I'm not sure how to implement it in this case. If you can help me it would be great.
Lucas

phil-barrett, 2022-12-26:
By 5V differential, I think that means 2 wires per signal. when one wire is +5V, the other is 0V.  (and not -5V). Looking through the manual, I'm not sure which inputs are step and dir - are they the inputs in section 4.3?  It looks like they are talking about the encoder pulse stream.

tonluc123, 2022-12-26:
Yes they are the inputs described at section 4.3. This manual looks like it was written for rocket scientists, I was fighting for hours to understand what the author meant to say. Anyway those inputs are not for the encoder connection. For 5v the step inputs are #1&2 and for dir #26&27. If you want I can send you photos by mail.

tonluc123, 2022-12-26:
Phil,
I just found this https://cache.industry.siemens.com/dl/files/415/109757415/img_249458/v1/2_en.png
Hope it helps a little

phil-barrett, 2022-12-27:
OK, makes more sense now.  I see 2 approaches:

1. 5V differential - use 1, 2, 26 and 27.  Use inverting drivers to create differential from single ended 5V output from T41U5XBB.
2. Create single ended 24V from 5V single ended.

Either one should work.  #1 may be simpler in that you will not need to supply 24V.

For #1, I would use an inverting line driver.  The [MC74VHC540](https://www.mouser.com/datasheet/2/308/MC74VHC540_D-1811369.pdf) is an 8 channel inverting with high drive capability and 3.7 nS switching time.  For each single ended signal, I would input it to a channel (say A1) and then connect Y1 (the output) to A2 to derive the differential signal.  So, Y1 and Y2 are your differential signals to feed into the V90.  Repeat that for each additional signal.  Each 74VHC540 will allow you to do 4 singled ended signals.  This should work up to about 200 kHz step rate.

For #2, use a ULN2003 to boost to 24V.  Since the '2003 is open collector, I would connect to output to the ground level of the v90 input and +24V to the signal input.  The '2003 will pull the voltage low via the signal's ground pin. 

Let me know if you need me to draw up the circuits.

tonluc123, 2022-12-27:
Hi Phil,
I feel a bit stupid, I read those 2 posts but I don't se how are they related with my case (my English maybe). I liked though the solution of MC74VHC540 and even with a ULN2003 (easier to find) . A circuit would be great if it is not too much for you.
Lucas

phil-barrett, 2022-12-27:
Here's what I worked up. Note that this is for a single servo.  Repeat for each additional servo.  Note that common ground is NOT needed. In fact common ground will defeat isolation. 

The single to differential conversion speed max is probably greater than 1 MHz (unlike what I said earlier).  5V to 24V speed is around 1 MHz.

I would prototype with this first. Let me know how it goes.  I am thinking of creating a board that people can get made.
[image]

tonluc123, 2022-12-28:
This is great Phill, I have a minor question about ULN2003, how do you power it ? 
From my side I'm thinking for an easier solution with a BS170 (24v driver), https://www.onsemi.com/pdf/datasheet/mmbf170-d.pdf it has 10ns turn on & turn off time. I used an oscilloscope at T41U5XBB yesterday at normal speeds and I had 10 μs pulses. I believe it will work. Of course the solutions you propose are much better and the case of 74lvt240 much more professional.
Did you check this https://cache.industry.siemens.com/dl/files/415/109757415/img_249458/v1/2_en.png by the way? What do you think, is it differential or common ground ?

phil-barrett, 2022-12-28:
You can use individual FETs (or bipolar transistors).  The ULN2003 is actually made up of transistors. Each channel is a pair of transistors in a Darlington configuration. See the diagram below.  I like it because it is simple to use and has some protections (like against start up glitches). If you use the FET approach, be sure to pull the input low with a 5 to 10K resistor to prevent glitches at start up. Also, limit the base current to something like 6 mA to be safe.
[image]
You don't power it in the sense of a conventional IC. It works similar to your FET in that the input signal turns on the darlington pair which connects the output pin to ground.  So you connect your "load" between +V and the output pin like so
[image]
In this case the load is an opto isolator (very likely what is in the V90).  When the input (or base) signal goes high, current will flow through the load (opto in this case).

The doc you referred to shows that differential needs a common ground so I would go with that.  But, it is silent about single ended so I would try without common ground first and if that doesn't work, then connect the ground.

tonluc123, 2022-12-29:
Hi Phill,
This is the result with BS170, I see some attenuation at the rising edge I drive the gate with 1k in series and 5k to the ground.
I would appreciate your opinion with this approach. 
[image]

phil-barrett, 2022-12-29:
That looks like it should work. Most signals are not ideal and it depends on the capacitive load you are driving. You'll never see perfect square waves. Have you tested it with a load (either an opto, led or resistor)? See fig 11 in the datasheet. I would also put a small resistor (50 Ohm) in series with the drain to protect against shorts.  FETs can be fragile.

tonluc123, 2022-12-29:
And this is the ULN2003, the rising edge looks more or less the same with the one of Bs170
[image]

tonluc123, 2022-12-30:
Phill,
I have tested it with a resistor, as for perfect squares, your board outputs perfect squares (professional work, congratulations). 
I may be able to improve the output as I solder the drive on a prototyping board and non shielded cables.
In any case thanks a lot for your help with this (as it was not really an issue) and keep the good work.
Happy new year
Lucas

closed 2023-01-24 by tonluc123