phil-barrett/grblHAL-teensy-4.x #74: Some help please

tonluc123, 2023-01-23:
Hi Phill,
I obtained Eth. kit for Teensy (MagJack) and tried with a "naked T4.1 with no success. My hex file was build from: http://svn.io-engineering.com:8080/ with tall the network options. After reading some articles I was confused and decide to ask for some help.
P.S. With ## Teensy 4.1 ethernet :: https://forum.pjrc.com/threads/59776-Ethernet-and-lwip , and an example I can see that hardware is in order and I have feedback to my serial monitor.
Your help is appreciated.
Lucas

phil-barrett, 2023-01-23:
What Eth kit did you get?  mine or from PJRC?  

Also, I don't quite follow your naked T4.1 reference.  How are you hooking up the magjack if it is not on the breakout board?

tonluc123, 2023-01-23:
Hi Phill,
I got the the PJRC  kit. Naked means not at your board, it is connected with a ribbon cable (teensy to eth board).
No I have not tried the prebuild ones , I will in a short while
Lucas

phil-barrett, 2023-01-23:
OK, let me know what you find.  Also FIY, the PJRC kit will not work with the T41U5XBB - the magjack pinout is wrong.

tonluc123, 2023-01-23:
Just tried, no luck. I thought that you use the same rj45 connector with pjrc. Is it possible to find one of yours in Europe ?

phil-barrett, 2023-01-23:
Hmmm, do you have the Ethernet Teensy 4.1? There are 2 versions - one without ethernet.  You can tell by looking at the 6 pin 2mm header - there should be a small chip between the header and the MXRT1062 chip. 

As long as you use the PJRC board/cable combo and connect it correctly, it should work. Maybe check cable orientation? How did you verify the HW earlier?

tonluc123, 2023-01-24:
Yes it is an Ethernet Teensy 4.1. As I mentioned before I can verify that everything works fine by testing it with library's examples. Most probably It has to do with mistakes I do of how to apply the code. If everything that I need is included at the hex file that I downloaded from http://svn.io-engineering.com:8080/  and no further action is needed it should work. Please forgive me, I'm not a networks specialist maybe I miss something basic that I'm not aware of.

phil-barrett, 2023-01-24:
Ok, I think the problem is not with the ethernet support but maybe with your io builder options.  Try this json file.  Load it into the builder and build.  I was able to load the resulting .hex file.
[iMXRT1062_T41U5XBB.zip](https://github.com/phil-barrett/grblHAL-teensy-4.x/files/10492571/iMXRT1062_T41U5XBB.zip)

tonluc123, 2023-01-24:
How do I flash the json file ?

tonluc123, 2023-01-24:
Phill I just managed to connect: used static ip instead of dhcp. Can you understand why is this ?
Lucas

tonluc123, 2023-01-24:
Can I send you pm ?

phil-barrett, 2023-01-24:
Use email - phillip dot lyne dot barrett at gmail.com

phil-barrett, 2023-01-24:
If you are connected, go into Grbl settings in ioSender.  Under network select dynamic for dhcp.  Not sure why your build was set to static.  Mine was set to dynamic.  Could have been an earlier setting.

tonluc123, 2023-01-24:
Thanks for been so helpful

closed 2023-01-24 by tonluc123