phil-barrett/grblHAL-teensy-4.x #75: Wiring XYZ Probe

hosquiat, 2023-03-11:
I'm looking to use the XYZ Probe Plus by OpenBuilds as a touch probe? I see they're site says it is compatible with GRBL and was looking for more information on how to wire the optional V+ input. 

[Spec](https://docs.openbuilds.com/doku.php?id=docs:xyzprobe:start) 

[Wiring Docs](https://docs.openbuilds.com/doku.php?id=docs:xyzprobe:wiringgeneric) 

[image]

Should I be using an external voltage source?

phil-barrett, 2023-03-11:
I looked to see what kind current it draws and they don't seem to list it.  (They really should say. )  However, It looks like all it does is light an LED so the power consumption is probably very small.  You could use power from the T41U5XBB.  The 5V terminal near the probe terminal, if open, would be fine.  External 5V is also OK as long as you connect the PSU Ground/0V to the T41U5XBB.

hosquiat, 2023-03-11:
Cool. I'll test this out once I get my machine up and running. Thanks for the fast response.

closed 2023-03-11 by hosquiat