phil-barrett/grblHAL-teensy-4.x #81: Teensy 4.1 DEV-16771 + lightburn

distebia, 2023-09-21:
Salve, ho intenzione di passare da mks dlc32 esp32 a Teensy 4.1 DEV-16771 e usare grblhal...posso continuare ad usare la licenza gcode di lightburn oppure devo aggiornare la licenza?

phil-barrett, 2023-09-21:
Please forgive my lack of Italian...

I have an older version of Lightburn but believe all you have to do is click on the devices button. Try Create Manually and select Grbl.  I think you can test it right now.

distebia, 2023-09-21:
salve e grazie per la risposta...non ho ancora comprato la scheda T41U5XBB e aggiungere il Teensy 4.1, non so come muovermi in quanto non trovo la T41U5XBB in europa, ma mi piacerebbe entrare in questa avventura. la mia domanda era inerente alla versione di lightburn da utilizzare, la versione base per gcode oppure quella dsp?

phil-barrett, 2023-09-21:
Ah, yes.  The Basic Version.

You can purchase the T41U5XBB from  [tindie.com](https://www.tindie.com/stores/philba/).

distebia, 2023-09-21:
ok per la versione. Avevo visto il sito e ho anche chiesto quale era la differenza di prestazioni, per uso laser co2, non cnc con mandrino ed interferenze) fra la versione usb e ethernet. inoltre il Teensy 4.1 non è fornito nel kit, vero? esiste un modo per comprare il prodotto da un negozio in europa? le spese doganali incidono tanto, oltre al trasporto

phil-barrett, 2023-09-21:
Ethernet vs USB - Ethernet is same speed as USB.  Ethernet has isolation for anti-EMI.  But, Lightburn Version 0.94 does not support Ethernet for Grbl, only USB.  I do not think they have changed that for Version 1.0 and later.

Tindie is the only source for T41U5XBB.  Sorry for the expensive shipping, I can not do anything about that.

distebia, 2023-09-22:
ok, mi sono quasi convinto per l'acquisto della sola scheda, kit base di $ 29.99, solo che mi serve sapere con esattezza cosa devo aggiungere, magari avere il termine tecnico per trovare il giusto connettore e pin a saldare. Altra domanda...l'alimentazione della scheda avviene tramite cavo usb, quindi dal pc, del Teensy 4.1? La mia configurazione (autocostruita) di 1300x1000 attualmente è composta da mks dlc32 che comanda 3 tb6600 che comandano 3 17HS8401S. Tutto scorre su guide lineari di 20mm con relativi pattini, nessun problema qua.
Passando al VS progetto avrò problemi con la mia meccanica, nel senso che i motori o i driver sono pensati per una macchina + lenta?

phil-barrett, 2023-09-22:
You can power it from USB, a USB charger (if you use ethernet) or external 5V. If you want to use 0-10V to control a spindle or drive 12V relay coils, you will need to supply 12V also.

You should probably take a look at ordering the USB assembly kit as that has all the connectors you will need. Make it very easy to get up and running. If you do not mind ordering the connector parts separately yourself, the user manual has a bill of materials for it.

I do not see any problem converting over - your mechanics sound fine. You can make the board run at the speed that they need.  You should write down all of your Mach3 settings before switching.

phil

distebia, 2023-09-22:
ho trovato online un teensy 4.1 completo di pin saldati ai lati (24+24) e la connessione 2x3 2mm ethernet gia saldato nel teensy 4.1. la differenza di kit da usb a ethernet è il connettore ethernet + il pin 2x3 2mm? oppure qualche altra cosa? circa l'alimentazione non trovo niente online. attualmente utilizzo relè 3.3v...devo usare 5v? dopo l'acquisto posso chiedere eventuali dubbi per il montaggio, firmware e altro?

phil-barrett, 2023-09-22:
The Ethernet kit has the MagJack, 2x3 2mm header and 2x3 2mm socket.  Be aware that if you buy a MagJack separately, you need to get the exact part number. There is no standardization.

The 2x3 2mm header will have to match the socket.  I think it will be ok but the ethernet pins need to be the same length as the other 24 pins on the sides of the teensy.  Could you send me a link?  Soldering 2mm pins is a lot harder than 2.54mm for some of my customers.

distebia, 2023-09-22:
ok per il MagJack...che succede se inserisco il Teensy 4.1 e saldo i 24+24 + i 2x3 2mm e non connetto il magjack? non saldo completamente il MagJack. Non saldando il connettore ethernet semplicemente non ho ethernet o, magari, il jak chiude qualche circuito?

phil-barrett, 2023-09-22:
A bad connection will not cause any problems other than no Ethernet.

distebia, 2023-09-23:
scheda comprata, grazie per i consigli...ci risentiamo quando arriva la scheda e il teensy per l'installazione di grblhal. Grazie

distebia, 2023-10-01:
Salve, per collegare un sensore pnp no con 5v-s-g 
[image]
posso collegarlo quà?
[image]

distebia, 2023-10-01:
Altro quesito: per collegare un relay devo usare necessariamente 5v? non posso usare 3.3v come per mks dlc32 (esp32).
Per un 5v immagino di collegarlo in questo modo:
[image]
Fornisco esternamente 5v nel connettore in basso e alimento tutta la scheda, mettiamo 5v 1A?, imposto il selettore 5-12v su 5v mediante jumper, prendo un qualsiasi G e lo porto al G relay, prendo dalla sezione Mist +V e lo porto a V del relay e S dal relay lo posto a Mist? 
Inoltre alimentando esternamente la scheda penso che devo modificare il cavo usb che va al pc..interrompo nel cavo il 5V ottenendo un cavo usb con 3 fili (-D, +D e G)

phil-barrett, 2023-10-01:
For the PNP sensor.  The connections you show are ok but I have not tested it with PNP sensors.  By the way, optical sensors like you show can be false triggered by debris (saw dust? swarf?  I don't know the Italian word for that).  

On relays, no 3.3V option is available.  What you wrote works to drive actual relay coils.  You do not need to modify the USB cable.  You can cut the USB solder jumper trace on the Teensy or cut the solder jumper trace on the bottom of the T41U5XBB.  I recommend cutting the Teensy.  But, If you are using powered Relay modules, you should use the TTL outputs instead of the relay outputs. The relay outputs are for directly driving relay coils.

distebia, 2023-10-01:
[image]
dentro ho sia relè 3.3v che 5v come questo. userò il 5v. suppongo che mediante il pin S non vado ad alimentare direttamente la bobina...in questo caso devo usare le uscite ttl e non relè? quindi S dal modulo relè lo collego qua?  
[image]
Ma il .hex generato da web build prevede Mist su questo pin? o devo modificarlo ricompilando?
Circa i PNP quando impegnato ho 5V su S, invertendo il $5 risolvo? 
Non mi va di rovinare la scheda tagliando tracce e ponticelli, L'alimentazione è esterna e non mi serve da usb...dovrebbe funzionare lo stesso...o mi sbaglio? se sbaglio posso sapere quale traccia tagliare?

phil-barrett, 2023-10-01:
You are on the right track but use the Gnd (G) next to the Signal (sig).  
[image]

phil-barrett, 2023-10-01:
scusatemi, avrei dovuto postarlo prima.
https://www.grbl.org/single-post/using-relays-with-grblhal-and-the-t41u5xbb

distebia, 2023-10-01:
Grazie per la guida dei relè, ben fatta. Per quanto riguarda  i PNP posso usarli invertendo la logica di $5 ? In questo momento alimento la mia mks dlc32 con 12v e, per non stravolgere l'elettronica, posso alimentare la T41U5XBB con 12V, commutare il jumper su 5V per usare i relay a 5V con le connessioni indicate? 
[image]
 Inoltre il pin da usare per comandare un laser, Co2 oppure diodo, è quello indicato in figura?

phil-barrett, 2023-10-01:
As to PNP, I do not think PNP will work with the T41U5XBB. I recommend NPN Open Collector drivers.   The Limit, Probe and Control inputs are activated by connecting the signal input to ground which is what an NPN driver does.

On your Laser question.  It depends on your Laser module but in most cases, a 5V PWM signal will work.  Check your Laser manual.

distebia, 2023-10-02:
Per i PNP, oltre a cambiare i sensori, esiste qualche convertitore logico da pnp a npn in commercio? Per l'alimentazione a 12V posso fare come descritto sopra? Circa il PWM i miei laser accettano 0-5v PWM, quindi dovrebbe andare bene

phil-barrett, 2023-10-02:
I do not know of any product that does the conversion. I looked on the internet and only saw wildly expensive offerings that cost more than getting NPN sensors. However, it is a very simple concept, just use an NPN transistor that takes input from the PNP output. Very cheap to make if you have some electronics skills.

On the laser, you may need to play with the PWM frequency though I bet the default of 5kHz will be fine.

distebia, 2023-10-02:
Grazie, perfetto. Dal NPN out (2) vado a pin X e 3 (gnd) a G sul pcb? Grazie ancora

phil-barrett, 2023-10-02:
yes. 2 to signal, 3 to gnd. Any small signal NPN will work.

distebia, 2023-10-03:
> Grazie per la guida dei relè, ben fatta. Per quanto riguarda i PNP posso usarli invertendo la logica di $5 ? In questo momento alimento la mia mks dlc32 con 12v e, per non stravolgere l'elettronica, [...]
dimenticavo la cosa + importante...circa l'alimentazione è corretto cosa affermo per i 12V? da qualsiasi controllo relè posso prelevare 5V per alimantare anche i finecorsa? la pista da tagliare è questa mostrata in figura? 
[image]
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              ,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

phil-barrett, 2023-10-03:
Do not use the bottom 5-12 solder jumper.   You do not need to cut any traces for that. It is the same as the top pin header jumpers.  I will be removing the bottom one in the next version of the board.

You can use 5V from the board but be aware that it is supplied by the USB connection.   I would try it with the relays. If it works with no problems, then go forward with it. However, if it creates disruption (EMI or voltage fluctuations can cause grblHAL misfunction) you may have to supply +5V from a separate power supply unit (PSU).  I hope that translates well.

distebia, 2023-10-03:
ripropongo la domanda, forse non sono abbastanza bravo a farmi capire...alimentazione da usb ok ma preferisco avere la macchina indipendente dal pc. Per questo motivo vorrei alimentare la scheda con 12V che già ho. Collego l'alimentazione qua

[image]

dopodichè imposto il jumper su 5V e su tutte le uscite relè ho 5V.

[image]

quindi collego il mio relè prelevando +5 dalla parte relè, S e G dal connettore in basso

[image]

Quindi presumo che fornendo 12V e settando il jumper su 5V i 12V incontrano qualche lm7805 che fornisce 5V sui relè. Se così non è allora ho capito male, mi scuso per questo. 

Se anzichè fornire 12v vado a fornire 5V li fornisco dal connettore in basso, setto il jumper su 5V e ho i +5v sui relè e tutto procede regolarmente.

[image]

Ma in questa situazione devo tagliare la traccia sotto, vero? La traccia è questa? 

[image]

Grazie

phil-barrett, 2023-10-03:
There is no 12 to 5V voltage regulator on the T41U5XBB.  The 5V comes from the USB connection to the Teensy.  You could use a USB charger for that, assuming you are using Ethernet to connect to grblHAL.

If you cut the jumper for ext 5V (as shown in your last image), you will need to provide +5V on the +5V screw terminal.

distebia, 2023-10-13:
> Probabilmente qualcosa del genere. Nessuno dei due resistori è assolutamente necessario. Il resistore di pulldown forma un carico per garantire che lo spegnimento e l'accensione siano unici, potrebbe essercene già uno nel circuito PNP. Il 1K serve a limitare la corrente. Ho indovinato i valori dei resistori: non sono critici. [image]

[image]

[image]
Salve, dentro ho 2 moduli da 4 uscite come queste i foto...potrebbe andare bene invece di collegare i transistor npn?

phil-barrett, 2023-10-13:
That may work.  Though, the 3K ohm resistor may limit the current to be too low for the opto isolators on the T41U5XBB. If you have one, try it.  If it doesn't work, you could try shorting out the resistors - making it 0 ohms. You won't hurt the T41U5XBB because the limit inputs have resistors built in.  I've circled the resistors.
[image]
You should remove the jumpers.

phil-barrett, 2023-10-16:
I bought one of those boards (HY-M154) and tested it with the T41U5XBB.  It works.  I was able to connect it to the limit inputs and could see it triggering in ioSender.  I used 12V and connected an NPN prox sensor.  I do not have an PNP sensor but I am sure it will work. 

Please forgive the messy wires.
[image]

distebia, 2023-10-16:
perfetto...in entrata del pc817 posso anche inviare 12v dal sensore...come da consiglio ho escluso r di la 3K mediante ponticello...funziona da te con 3k di resistenza?

phil-barrett, 2023-10-16:
Yes, it works with the resistor in place.

distebia, 2023-10-17:
Se devo configurare 2 endstop per asse, quindi +X, +Y e +Z dove devo aspettarmi l'ingresso del finecorsa?

phil-barrett, 2023-10-19:
The User Manual will help you.  You can get it from the [Resources Page](https://www.grbl.org/resources).

distebia, 2023-10-19:
si ho letto ma ancora non riesco a capire dove termina il pin del teensy nella T41U5XBB. ho aggiunto il doppio endstop sui 3 assi. 

[my_machine_map.h.txt](https://github.com/phil-barrett/grblHAL-teensy-4.x/files/13040920/my_machine_map.h.txt)

non capisco dove terminano i pin 23, 28 e 29 da usare per +x, +y e +z. sarebbe bello avere 
[image]

con le varie uscite pin.

phil-barrett, 2023-10-19:
[image]

distebia, 2023-10-19:
[image]

quindi collego i finecorsa in questo modo? -X su X (20u), +X su A (23u), -Y su Y (21u), +Y su B (28u), -Z su Z (22u), +Z su Door/Sig (29u)  ?? 
Oggi ho iniziato ad assemblare la scheda con i vari pin, vediamo sosa esce fuori, posterò le foto. nel frattempo ho provato a compilare con Platformio e anche con Arduino ide ma ricevo errori...in questo post ho ricevuto parziale aiuto e non capisco se sono errori o semplici avvisi  
https://github.com/grblHAL/iMXRT1062/issues/69

phil-barrett, 2023-10-24:
OK, I think I misunderstood.  For a given axis with one switch, you connect + to signal pin and - to the G pin. (for X axis, + to X, - to G).  

Or, are you talking about 2 limit switches per axis?

distebia, 2023-10-24:
certo che parlo di 2 finecorsa per asse... quindi collego i finecorsa in questo modo? -X su X (20u), +X su A (23u), -Y su Y (21u), +Y su B (28u), -Z su Z (22u), +Z su Porta/Sig (29u) ? ? ma dove si trovano fisicamente 23u, 28u e 29u sulla T41U5XBB?

distebia, 2023-10-24:
ritornando sul discorso relè mi chiedevo se è possibile alimentare un relè logico ttl 3v in questo modo
[image]

phil-barrett, 2023-10-24:
No, TTL/SSR output is 5V only.  You could set up a voltage divider to reduce the voltage to 3.3V though.

distebia, 2023-10-24:
ok capito per i relè, monto quello a 5v. circa l'altra domanda? 

> certo che parlo di 2 finecorsa per asse... quindi collego i finecorsa in questo modo? -X su X (20u), +X su A (23u), -Y su Y (21u), +Y su B (28u), -Z su Z (22u), +Z su Porta/Sig (29u) ? ? ma dove si trovano fisicamente 23u, 28u e 29u sul T41U5XBB?

distebia, 2023-10-25:
[image]
Salve, so che il connettore ethernet deve essere uno di quelli indicati nel manuale ma se ne monto uno con piedinatura uguale funziona? oppure deve avere anche le specifiche elettriche come da immagine allegata?

phil-barrett, 2023-10-27:
If yo find one with exactly the same pinout, it should work but I have only tested with the one in the bill of materials (BOM).

On the sensor/switch front.  I have 3/4 of the answer for you.  You connect min and max sensors for given axis together and into the single limit input for that axis.  

How you do it depends on whether they are NO or NC.  NO are wired in parallel.  NC are wired in series. 

If you have prox sensors, NO can be wired in parallel.  The part I do not know the answer to is NC prox switches.  I believe you can "stack" them but would want to test it out to see.

distebia, 2023-10-27:
grazie per la risposta...in fluidnc assegno un determinato pin a qualsiasi cosa mi serve. con grblhal ho fatto la stessa cosa, ho modificato my_machine.h e my_machine.map.h assegnando i pin come in figura

[image]

dove trovo i pin corrispondenti sulla scheda?
ho anche intenzione di mettere 2 endstop al limite sotto e al limite sopra per l'asse Z. L'asse Z non ha homing. Sulla testa laser ho un induttivo che rileva il piano che sale (cambiando materiale cambia il mio fuoco) e mi posiziona l'asse Z a una certa distanza. Non voglio attivare l'hard limit. Supponendo un guasto del sensore induttivo, l'asse continua a salire rovinando tutto. Dovrei configurare i 2 endstop come comandi di stop immediato, chiamalo Halt, Door, l'importante che mi blocca immediatamente la salita oltre il finecorsa induttivo guasto. 

per quanto riguarda il connettore ethernet rj45 ne ho trovato uno (da recupero) identico per piedinatura tranne per la parte led (ho saldato 2 led sotto il pcb rispettando le polarità). al connettore mancanza il filtro/trasformatore interno. Dopo aver generato il firmware con ethernet ho provato a fare la configurazione (da guida) ma non sono riuscito a visualizzare nessun indirizzo ip. 
Il filtro influisce soltanto alla riduzione dei disturbi oppure è una parte fondamentale per il funzionamento del T41U5XBB?

phil-barrett, 2023-10-27:
Sorry, that is an incorrect board map.  The Web Builder will give you the correct one. Select Teensy and T41U5XBB and then press Board Map (circled in red).
[image]

The Ethernet Mag Jack is driven by the Physical Link controller (Phy) on the Teensy - TI dp83825.  Take a look at the datasheet for more specifics.  It shows several configurations.  I referred to it when designing the Ethernet support.  Ethernet Magnetics are a bit of "Voodoo". By the way, make sure you have a version of the Teensy that has the Phy chip on it.  There is a version that does not have it (Teensy 4.1 NE).
https://www.ti.com/lit/ds/symlink/dp83825i.pdf?ts=1698412030686&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FDP83825I

distebia, 2023-10-27:
https://www.amazon.it/dp/B0BP5BNC9K?psc=1&ref=ppx_yo2ov_dt_b_product_details

questo è il teensy che ho comprato...parla di ethernet. ho saldato i 6 pin e ho testato la continuità, entrano perfettamente su T41U5XBB e ho continuità. l'unica cosa che ho notato è che il led del rj45 non lampeggia e che il fritzbox non riconosce niente attaccato con il cavo ethernet. domani allego le foto reali del teensy

phil-barrett, 2023-10-28:
That has the ethernet Phy chip on it - right next to the 2x3 2mm header.

distebia, 2023-10-28:
il teensy ha il chip, l'intestazione 2x3 2mm è ok, il socket sotto è ok, il connettore è uguale, il firmware è ok (almeno credo)...cosa è andato storto?

phil-barrett, 2023-10-28:
When you plug the ethernet cable in to a router/switch, does the port light turn on? Also, one of the LEDs should start flickering when connected to a working ethernet.

On the other hand, perhaps you could find the RB1-125BAG1A magjack in the EU? They are fairly common and not terribly expensive.

distebia, 2023-10-28:
I led non si accendevano e non lampeggiavano, comunque ho smontato il connettore e messo da parte. Ho notato invece che alla fine del lavoro il controller rimane in stato occupato. La macchina è sempre la stessa e prima con mks dlc32 non succedeva. Succede anche che quando parte la lavorazione si blocca senza segnalare errore e devo spegnere. All'inizio del lavoro viene acceso il relay ttl e spento alla fine del lavoro, forse è questo? Ho rigenerato il firmware 3 assi solo con LB cluster attivato e devo ancora testare.

phil-barrett, 2023-10-28:
No LED lit means the phy has not been able to establish connection to the ethernet router/switch.

distebia, 2023-10-28:
Devo riprovare saldando nuovamente il connettore. E per i problemi di "stato occupato"?

phil-barrett, 2023-10-28:
I have never encountered busy status so not at all sure what it means.

distebia, 2023-10-28:
[image]
questo è lo stato occupato, non risponde ai comandi e devo spegnere e riaccendere

phil-barrett, 2023-10-28:
Ah, Lightburn.  I'm not sure what that means.

distebia, 2023-10-28:
Si, sono collegato tramite usb. Lo stato busy si verifica all'inizio e alla fine del lavoro, ma non sempre. il controller va in stallo e non accetta comandi, devo togliere l'alimentazione e riaccendere. Altre volte basta uno stop e tutto riparte senza perdere la posizione. questo problema mi succedeva con arduino, poi ho montato anelli di ferrite sui segnali step e dir che dal controller vanno ai tb6600 e le cose sono migliorate. ho anche aggiunto dei condensatori 275vac x2 (su fase e neutro) di alimentatore laser co2, pompa acqua e compressore aria. con arduino non ho avuto più problemi, sono passato a mks dlc32 e non avevo problemi del genere. Per quanto riguarda l'alimentazione fornisco 5v sui pin della T41U5XBB. Non ho interrotto tracce nel pcb. Utilizzo un cavo usb senza i 5v e funziona. non capisco cosa può essere. Quale altra precauzione posso prendere per evitare questi blocchi?

distebia, 2023-10-29:
ho superato i problemi di stato occupato...era bruciato un condensatore 275vac x2 messo nella pompa dell'aria, infatti mi busy mi succedeva all'inizio e alla fine del lavoro. Ho anche riscontrato problemi con l'alimentazione 5v che fornivo alla scheda, mi mandava in tilt casuale il teensy. attualmente non fornisco alimentazione alla scheda e alimento solo con usb...tutto funziona. 
quello che non funziona è la connessione ethernet, non ottengo l'ip e i led non lampeggiano. fra qualche settimana mi dovrebbe arrivate un RB1-125BAG1A non trasformatore interno...

distebia, 2023-11-09:
Purtroppo ritorno al problema dell'alimentazione da scheda...con qualsiasi cosa (lm7805 1.5A +condensatori, alimentatore 5v stabilizzato, alimentatore per ricarica cellulare 5v 2a) il teensy va in stallo durante il lavoro, si congela e non accetta niente, posso solo spegnere. Se alimento da usb tutto perfetto. tutto questo mi succede dopo aver caricato un firmware compilato da me (platformio e senza errori o avvisi). Se carico un firmware generato da web build il problema scompare (almeno credo). Posso riutilizzare i miei alimentatori per fornire 5v alla scheda e cavo usb solo dati verso pc. Mi sembra strano ma è così. Allora penso a qualcosa di sporco che carico io nel teensy, o qualcosa che rimane sporco dopo il caricamento del firmware. I problemi sono iniziati quando ho variato il $398=35 e l'ho modificato a $398=100, dopo questa operazione il teensy non digeriva bene l'alimentazione 5v da scheda. Come posso formattare completamente il teensy? Carico lo schizzo blink e poi il firmware generato da me?