phil-barrett/grblHAL-teensy-4.x #84: Flash Firmware

distebia, 2023-10-05:
Salve, ho caricato il firmware generato da web build, tutto perfetto e velocissimo. in mente persiste una domanda: mettiamo che il firmware caricato ha bisogno di una variazione per adattarsi alla macchina, genero nuovamente il firmware e lo carico ma devo cancellare quello esistente oppure la procedura è automatica? In sostanza devo formattare il teensy 4.1 prima del nuovo flash?

phil-barrett, 2023-10-05:
No, just upload the newer firmware. It will preserve your previous settings.

distebia, 2023-10-05:
ma se per qualche motivo devo riportarlo a zero e ripartire? esiste tipo un comando erase_flash?

phil-barrett, 2023-10-05:
Teensy.exe will erase before programming.

If you want to reset grblHAL, *rst will reset all settings back to default values.

distebia, 2025-01-16:
Salve, grblhal ha implementato la possibilità di avere + mandrini (anche pwm) attivi allo stesso tempo, o magari attivi uno alla volta...su teensy 4.1 e T41U5XBB quale potrebbe essere un valido pin in grado di erogare un pwm 0-5V?

phil-barrett, 2025-01-17:
Many of the pins on the teensy are capable of 5V PWM. I don't believe you can use them via the Web Builder, though. You would have to change source code and recompile. In addition,  the T41U5XBB will only support pins set up for output - top and right edges on the board.