phil-barrett/grblHAL-teensy-4.x #86: Direct ethernet link

LKG-atelier, 2023-10-29:
Hi,

I would like to link the T41U5XBB via direct ethernet to the computer (linuxlite based).
Do you think that's possible ? 
Here it talks about an ethernet switch or hub (https://www.grbl.org/single-post/networking-the-t41u5xbb)
but would it be possible without ?
(I've tried, but networking configuration is a bit harsh on linux...)

Thanks for your replies
Best regards
Luc

phil-barrett, 2023-10-29:
Yup!  [This](https://www.grbl.org/single-post/ethernet-without-routers-for-grblhal) is focused mainly on windows but you can use ifconfig to set up the network on linux. You will have to identify the network interface to use on your machine.  The grblHAL side is super easy, basically just one setting to change  - $301, IP mode.  send $301=0.

LKG-atelier, 2023-11-03:
Thank you very much
it works like a charm !

closed 2023-11-03 by LKG-atelier