phil-barrett/grblHAL-teensy-4.x #87: T41U5XBB network connection

Hatty99, 2023-11-01:
Hi Phil,
I'm attempting to connect myT41U5XBB  board to my network. 
Have generated and loaded firmware as per your instructions: https://www.grbl.org/single-post/networking-the-t41u5xbb, with Ethernet/telnet/Websocket all checked, and attempted to connect to the board via the network, unfortunately unsuccessfully. As far as I can tell USB connection is working OK. (machine homes and can move with gcode)
The board is connected to the network, and the network connector led is flashing. It successfully obtains an IP address, but there is no message indicating a NETCON connection. I can ping the ip address from my Windows laptop, but am unable to open a connection using telnet from the laptop.
When starting ioSender, the program shows the message "Unable to open connection (ws://192.168.XXX.XXX:80)" showing the correct IP address.
Could you be able to either point me in the right direction or suggest any further diagnostic moves?
Thanks,
Dave.

phil-barrett, 2023-11-01:
Did you close the USB connection?

Hatty99, 2023-11-01:
I believe so. Have disconnected the USB cable and the com port has disappeared from Device Manager. 

When the com port is still open, ioSender connects via USB. When the port is closed I get the "Can't open connection" message.

5v and 3.3v on the teensy board seem ok without the usb connected.

Hatty99, 2023-11-01:
Thanks for your prompt reply and help Phil. Changing to port 81 has made it work.

closed 2025-03-05 by phil-barrett