phil-barrett/grblHAL-teensy-4.x #102: calo improvviso emissione laser

distebia, 2024-02-10:
Salve, oggi stavo facendo una lavorazione su 3 livelli colore di lightburn, 1° livello foto modalità stucki, 2° riempimento di contorni e 3° taglio. Il problema si presenta dopo aver fatto perfettamente la foto, finisce la foto e passa al riempimento. Effettua si e no 20 linee di riempimento, impostato a 8000mm/min 20% 0.08 linee per millimetro, e all'improvviso mi accorgo che il laser si muove ma non ho emissione di fascio, l'amperometro da 0-30mA segna forse 2 e nel plexiglass si vede che il riempimento non c'è. Blocco tutto, rifaccio l'homing, seleziono soltanto 2° livello-riempimento e 3° livello-taglio e porto a termine il lavoro. Utilizzo teensy4.1+T41U5XBB. Non riesco a capire da cosa può dipendere tutto questo, la macchina funzionava prima e funzionava anche dopo, è come se il controller abbassa, ma non azzera, l'uscita pwm, dopo l'homing (senza toccare altro, tipo spegnere e rieccendere ecc...) tutto funziona come prima. Anche la temperatura era nel range valido e segnava 13.4 °C

phil-barrett, 2024-02-10:
Sorry to hear of the problem.

Was there any message in the console?  Could you still talk to grblHAL? (for example, send $$, get settings listed back?)

What happens when you run the same job?

distebia, 2024-02-10:
potevo parlare con grblhal, ho premuto stop su LB e ho eseguito l'homing. Niente è stato spento per raffreddarsi, dopo homing ho ripreso subito il lavoro

phil-barrett, 2024-02-10:
OK, keep an eye on the Teensy LED if this happens again.

distebia, 2024-02-10:
ok, l'amperometro segnava, poco ma segnava, era come un'uscita settata al 20% ma eseguita al 1%.