phil-barrett/grblHAL-teensy-4.x #107: No IP Address found for T41U5XBB Ethernet connection

fudgtastic, 2024-05-03:
Hello,

I have been working through the "Networking the T41U5XBB" blog post: https://www.grbl.org/single-post/networking-the-t41u5xbb. I have got to the step 

_"In ioSender, open the Console tab.  You should see something similar to this.  Note the line circled in red. That shows the IP address of the T41U5XBB."_

I have run into an issue where my IP address in the console is listed as "[IP:]" (no IP address listed). I've tried poking around in the windows network settings, but I can't see anything in particular that indicates the ethernet port is not connecting. I don't really know what I am doing in this area, though, so a network setting may indeed be the issue.

Is there anything I should try to find this IP address?

Thank you for any support,
Drummond

[image]

fudgtastic, 2024-05-04:
I believe I have resolved my issue. The nature of the problem was that I was direct connecting my ethernet. I think in my own little context, my ethernet router is entirely on the other side of my house from my shop, so the thought of connecting the CNC controller to a router/modem/switch never crossed my mind. The thought was always to connect direct to the PC.

Pair that with some getting familiar with github (e.g. looking at closed issues) I realized that there is already a guide on direct-connecting via ethernet: https://www.grbl.org/single-post/ethernet-without-routers-for-grblhal

The guide was useful, straightforward, and seems to have worked great. 

I have left a comment on the grbl.org post to direct anyone like myself to the proper resource.

phil-barrett, 2024-05-04:
Glad you solved it!  And thanks for the comment.  I updated the blog to include a link to the Static IP Address blog post.

closed 2025-03-05 by phil-barrett