phil-barrett/grblHAL-teensy-4.x #109: Question about 5V power

JasonTitcomb, 2024-06-21:
Regarding T41U5XBB. v2.09
And the documentation here.
https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/T41U5XBB%20v207.pdf
Page16

I was planning on using the USB from the Teensy so it appears I need to cut the trace.
I also plan on using the 12 relay option and the jumper is set.

I tested that there is no continuity, but the LED next to the +5V screw terminal still lights up when I plug the USB into the Teensy. 
Nothing else is plugged into the board yet.
I have built two of these and they behave the same.
What could I be missing?

phil-barrett, 2024-06-21:
Cut the trace only if you are connecting +5V to the 5V screw terminal.  If you are using USB for connectivity, there is no need for an external 5V PSU.

If you are referring to the relay outputs on the screw terminals, they operate as open collector drivers.  [Here is a blog](https://www.grbl.org/single-post/using-relays-with-grblhal-and-the-t41u5xbb) on how to use them.  

[This blog](https://www.grbl.org/single-post/ttl-ssr-on-the-screw-terminals) talks about the open collector driver in some detail.

closed 2024-06-21 by JasonTitcomb

Perlovka, 2025-01-13:
After cutting the trace step motors are not running anymore, no matter if 5v or 12v is connected to screw terminal.
Motors gets enabled and I can hear sound from them, like they are trying to move, but no actual move on any axe.
Restoring trace immediately fixes the problem even when both 12v and 5v are connected.

phil-barrett, 2025-01-13:
Which trace are you cutting?  The one on the bottom of T41U5XBB?  If so, you then have to provide 5V to the boards (via the screw terminal).