phil-barrett/grblHAL-teensy-4.x #114: How to wire and configure for single_block

JasonTitcomb, 2024-10-12:
I am trying to understand how to add a toggle switch for single_block mode on the T41U5XBB.
$pins  in IO  Sender reports...
**[PIN:29 Aux in 6, Single_block]**
Can someone offer guidance?

phil-barrett, 2024-10-12:
I'm not sure how much I can help you.  Pin 29 is the Safety Door input on the T41U5XBB.  I've never tried single block mode, can you point me to any grblHAL literature on it?

JasonTitcomb, 2024-10-12:
This document lists pin 29 as the door pin.
https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/teensy%204%20pinouts.pdf
I will try that.

JasonTitcomb, 2024-10-12:
Hi Phil,
Using the door pins would be perfect for me.
I now see [your document](https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/teensy%204%20pinouts.pdf) stating this.

I am close to being able to try it.

It started with this issue.
https://github.com/grblHAL/core/discussions/422#discussioncomment-8062400
I am using the Web Builder and including the option "Single step G-code blocks".

phil-barrett, 2024-10-12:
Looks like you are on the cutting edge! Terje is on vacation so may not be super responsive.  Do you know how it is supposed to work? In reading up on single block for general CNC, it looks more like a mode where you hit Cycle Start, it executes one block and then goes into Feed Hold state. Is that what is supposed to happen here?

JasonTitcomb, 2024-10-12:
Yes

JasonTitcomb, 2024-10-17:
Single block works great so far.
Toggle switch is wired to the safety door by default and this works for me.
With single block enabled, each cycle start press executes just one line.
Then when confidence is high just switch single block off and cycle start to run normally.
If the job is already running and you toggle single block on then it seems to ignore it. This is fine with me.

phil-barrett, 2024-10-17:
Excellent!  I will have to play with it.  Thanks for the report.

JasonTitcomb, 2024-11-03:
Just an update to the single block feature.
It doesn't seem to honor the state of the physical switch when reset.

I cobbled together a test rig with a Arduino Due(_only because I can get the debugger working with that board_) and it seems to work correctly.
I thought it might be related to me using the safety door pin so I change. 
I enabled the safety door using the web builder and that forced it to use the  B limit.
I wired my switch to the B limit on the board and it still ignores the state of the pin upon start or reset.

This is not a big deal for me as I am just a hobbyist. Just trying to report my findings for testing.
FYI I have created another issue in the grblhal repo. [https://github.com/grblHAL/iMXRT1062/issues/87](url)

phil-barrett, 2024-11-03:
Yes, I noticed that you have to cycle the switch upon reset.  Should be reported in grblHAL/core issues.

JasonTitcomb, 2024-11-04:
I believe I found the problem and created a pull request.
https://github.com/grblHAL/iMXRT1062/pull/88

closed 2024-11-04 by JasonTitcomb