phil-barrett/grblHAL-teensy-4.x #116: RE: Drivers and 3.3v VS 5.0v and Other Ideas

silver2row, 2024-12-09:
Hello...

I have had my share of 5v drivers giving me trouble. Since the teensy4.1 controller is 3.3v, look for drivers that have at least 3.3v logic associated with them like Gecko Drivers. 

I got the drivers beforehand and tested them with the GRBLHal board and the Teensy4.1 add-on computer (32-bit).

Seth

P.S. This is sort of a plug but I am not associated with Gecko Drives. I just know they work and with ease because of their drivers being associated with 3.3v logic. Phew and done!

silver2row, 2024-12-09:
Also...

Is there talk of a newer board being produced? If so, I would like to get my hands on them... For instance, I say make the screw terminal headers larger for PWR and GND from the PSU to the GrblHAL board. 

Seth

P.S. Just two cents here. Nothing more.

phil-barrett, 2024-12-09:
Sorry to disappoint but the driver outputs from the Teensy (and Pico) Boards are 5V.  Many drivers do not work with 3.3V.

Other that your post, I have not had a single customer complain about driver problems that were due to 5V. Please share what problems you have had. I am always interested in how I can improve the products.

silver2row, 2024-12-10:
Oh...

I use the Teensy4.1 board on the breakout you made. I am not disappointed. I just cannot get the drivers for the DM332T to understand what it is I am commanding it to do. 

I got the Gecko Drivers to work, i.e. which is 3.3v to 0v min. and then 5v to 0v max. 

The machine is up and running now. I always knew of the Gecko Drivers working with your breakout and the Teensy4.1 and thankfully.

I was mistaken I presume. I thought the Teensy4.1 and `T41U5XBB`, without spindle sync, used 3.3v logic.

I have been racing up and down trying to find drivers to work with your the `T41U5XBB` and Teensy4.1 because I invested in the Teensy4.1 and would like to use your boards more. There is a GND, and I am completely guessing since most driver mfg. do not Open Source their builds, on the stepper headers along with STEP/DIRECTION. 

I thought the STEP/DIRECTION was a norm. It is. But...this darn driver, the DM332T, has this 5v OPTO pin that seems to give me issues. It is not them or you. It is me. I understand that the issue is mine. 

Anyway...here is a quick section of the DM332T manual to better associate what I am saying with actual tech. 

[image]

And sir, sorry to get on your bad side here. Just to repeat it, I am not disappointed with what you have built.

And yes sir, you are right. Most drivers do not work with 3.3v to 0v signals because of low switching if that is what the driver is catching.

Seth

P.S. Anyway, I understand the GND from the Stepper Screw Terminals on the `T41U5XBB` needs a way to associate a common GND with the Driver(s). The GND on the DM332T is not allocated to the pins (I/O). I am not trying to bust your bubble or ruin a good name of so-said drivers. I am just not getting why I cannot make the DM332T work with specified boards like the `T41U5XBB`. I got a Gecko Drive G251x to almost work. My shotty work on the soldering gave in...

Again. Sorry to be such a burden here. I got these drivers thinking it would be easy as cake. A cake walk. I know now it is me with the issue and I guess the support ends at that point. 

Anyway, I will post a video of my creation once it is building properly. I might have to change out a motor for the Y-Axis. It is a Tronxy and it is not reacting well to my current set up.

phil-barrett, 2024-12-10:
Oops, I was wrong, you won't need to invert $2 and $3.

silver2row, 2024-12-12:
https://youtu.be/O32DJEtwVCo?si=etbbW5ISsDA59x-h

silver2row, 2024-12-12:
I got some ideas finally working. Check it out!

phil-barrett, 2024-12-12:
Super glad to hear (and see) it!  Always happy to see people getting it working together.

Phil

silver2row, 2024-12-12:
Yeppers Sir,

The machine is small for now. I plan on building a bit larger machine but for now, that is it and yes. I am glad too!

Seth

P.S. I am going to step aside from the DM332T drivers for now, i.e. as I just do not know how to use them currently. So, onto other endeavors and I will stop this thread now. Thank you again, sir.

closed 2024-12-12 by silver2row