phil-barrett/grblHAL-teensy-4.x #117: Does ethernet time out?

stevempotter, 2024-12-20:
Hi Phil,
I am sometimes fiddling with my machine for some minutes before plugging in the ethernet cable and trying to connect to the sender. It seems the ethernet port is dead, no LEDs blinking. I have to power cycle the controller to get ethernet back, a pain if I have already homed and set (0,0,0) carefully with my pendant. 

Is it intended behavior that the ethernet port on Teensy gives up after some time? If so, can I disable that?

phil-barrett, 2024-12-20:
I've not seen that.  It may be in the networking libraries which would need a rebuild. You might want to ask on the [core github discussions](https://github.com/grblHAL/core/discussions) section. 

Have you tried a soft reset? (probably won't do it, though)

stevempotter, 2024-12-21:
I could not get soft reset (from my pendant) to work to fix the No Ethernet problem. I have posted my Q on the grblHAL/Core issues.