phil-barrett/grblHAL-teensy-4.x #118: Question about external power

oscmejia, 2025-01-29:
Hi all,

I'm building a router table with a Teensy 4.1 and the t41u5xbb Breakout board.
This is the second one, since I fired the first one. That's why I'm asking this time, before thying.

I'm reviewing the documentation for the breakout board, but I'm now clear about external power.
I have a PC connected to Teensy, and it provides 5v as expected.

1) I'm going to use inductive limit switches that require 5v, with only the usb connected, the 5v screw terminals do NOT provide 5v. I assumed it would by this in the documentation:

5V Input or Output
Supply for 5V logic and Teensy 4.1 when not power via microUSB connector.
Can supply upt to 150 mA to off board devices.

So, do I need to provide 5v external to the screw terminals and use the same screw terminals to power the 5v for the limit switches?
I think this is how I fried my first board

2) For the VFD/spindle, do I need to provide 12v external?

Thanks a lot!

oscmejia, 2025-01-31:
please help @phil-barrett

phil-barrett, 2025-01-31:
Sorry for taking so long to get back to you.

You are using USB to supply power to the Teensy?  This can supply external 5V via the screw terminal.  The limit noted is for that scenario.  The 5V inductive limit sensors should be well within that 150 mA limit.

12V input is required if you are using 0-10V for your spindle.

If you plan on using USB to communicate to the Teensy, you will not need an external +5V PSU.

oscmejia, 2025-01-31:
thanks Phil!
That's what I expected, but I'm not getting 5v from the terminals, when connected to the PC via USB.
Do I have to solder any bridge or similar?

phil-barrett, 2025-01-31:
How are you connecting them?  Have you modified the T41U5XBB?

oscmejia, 2025-01-31:
no, my T41U5XBB is not modified at all.
I'm just reading with a multimeter the 5v terminals and got nothing.
But yes, the LED for 5v lights up.

phil-barrett, 2025-01-31:
OK, so there is 5V power on the PCB but you get 0V on the 5V screw terminal?  Check the soldering on terminal.  The LED lighting up means that there is power on the PCB trace that goes to the +5V terminal. (make sure your multimeter is set to DC volts.)

oscmejia, 2025-06-04:
Hey @phil-barrett 
coming back to this. this is what I have:

https://ibb.co/gZynJtV5

Just want to make sure I should power in at the pins shown in the image, in order to use the spindle 0-10v pins, even if powering via USB.

thanks a lot.

oscmejia, 2025-06-08:
@phil-barrett please help me confirm this